JAndrassy/WiFiEspAT #1: problem with multiple connections

bialabs, 2019-10-02:
i am testing this library (1.0.1 and Espressif's AT firmware 1.7.1) with the following sketch:

```
#include "WiFiEspAT.h"

char ssid[] = "Wifi";            // your network SSID (name)
char pass[] = "pass";        // your network password
int status = WL_IDLE_STATUS;     // the Wifi radio's status

char server[] = "arduino.cc";

unsigned long lastConnectionTime = 0;         // last time you connected to the server, in milliseconds
const unsigned long postingInterval = 10000L; // delay between updates, in milliseconds

// Initialize the Ethernet client object
WiFiClient client;

void setup()
{
  // initialize serial for debugging
  Serial.begin(115200);
  // initialize serial for ESP module
  Serial3.begin(9600);
  // initialize ESP module
  WiFi.init(&Serial3);

  // check for the presence of the shield
  if (WiFi.status() == WL_NO_SHIELD) {
    Serial.println("WiFi shield not present");
    // don't continue
    while (true);
  }

  // attempt to connect to WiFi network
  while ( status != WL_CONNECTED) {
    Serial.print("Attempting to connect to WPA SSID: ");
    Serial.println(ssid);
    // Connect to WPA/WPA2 network
    status = WiFi.begin(ssid, pass);
  }

  Serial.println("You're connected to the network");
  
  printWifiStatus();
}

void loop()
{
  // if there's incoming data from the net connection send it out the serial port
  // this is for debugging purposes only
  while (client.available()) {
    char c = client.read();
    Serial.write(c);
  }

  // if 10 seconds have passed since your last connection,
  // then connect again and send data
  if (millis() - lastConnectionTime > postingInterval) {
    httpRequest();
  }
}

// this method makes a HTTP connection to the server
void httpRequest()
{
  Serial.println();
    
  // close any connection before send a new request
  // this will free the socket on the WiFi shield
  client.stop();

  // if there's a successful connection
  if (client.connect(server, 80)) {
    Serial.println("Connecting...");
    
    // send the HTTP PUT request
    client.println(F("GET /asciilogo.txt HTTP/1.1"));
    client.println(F("Host: arduino.cc"));
    client.println("Connection: close");
    client.println();

    // note the time that the connection was made
    lastConnectionTime = millis();
  }
  else {
    // if you couldn't make a connection
    Serial.println("Connection failed");
  }
}

void printWifiStatus()
{
  // print the SSID of the network you're attached to
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  // print your WiFi shield's IP address
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  // print the received signal strength
  long rssi = WiFi.RSSI();
  Serial.print("Signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
}
```

i have two problems
- sometimes (3 times on 10) i get "Connection failed" 
debug output:
```
esp> AT+CIPSTART=4,"TCP","www.mysite.com",80 ...sent
esp> ?
esp> busy p... ...ignored

esp> ?
esp> busy p... ...ignored

esp> ?
esp> busy p... ...ignored

esp> ?
esp> busy p... ...ignored

esp> ?
esp> busy p... ...ignored

esp> ?
esp> busy p... ...ignored

esp> ?
esp> busy p... ...ignored

esp> ?
esp> busy p... ...ignored

esp> ?
esp> busy p... ...ignored

esp> ?
esp> busy p... ...ignored

esp> ?
esp> busy p... ...ignored

esp> ?
esp> busy p... ...ignored

esp> ?
esp> busy p... ...ignored

esp> DNS Fail ...ignored

esp> ERROR ...error
esp ERROR: expected OK got ERROR
Connection failed
```

- sometimes (1 times on 10) Serial.write(c) doesn't  write anything (connect works in this case because i get "Connecting...") 
debug output:

```
esp> AT+CIPSTART=4,"TCP","www.mysite.com",80 ...sent
esp> 4,CONNECT ...ignored
esp> OK ...matched
Connecting...
esp INFO: send data on link 4
esp> AT+CIPSEND=4,64 ...sent
esp> OK ...ignored

esp> > ...matched
esp> Recv 64 bytes ...matched
esp> ?
esp> busy s... ...ignored

esp> ?
esp> busy s... ...ignored

esp> ?
esp> busy s... ...ignored

esp> ?
esp> busy s... ...ignored

esp> ?
esp> busy s... ...ignored

esp> ?
esp> busy s... ...ignored

esp> ?
esp> busy s... ...ignored

esp> 4,CLOSED ...processed
esp INFO: closed linkId 4
esp> SEND FAIL ...matched
esp ERROR: failed to send data
esp INFO: send data on link 4
esp ERROR: link is not connected.

esp INFO: close link 4
esp WARN: link is not active
esp INFO: free linkId 4
esp INFO: start TCP to www.mysite.com:80 on link 4
esp> AT+CIPSTART=4,"TCP","www.mysite.com",80 ...sent
esp> ?
esp> busy p... ...ignored

esp> ?
esp> busy p... ...ignored

esp> ?
esp> busy p... ...ignored

esp> ?
esp> busy p... ...ignored

esp> ?
esp> busy p... ...ignored

esp> ?
esp> busy p... ...ignored

esp> ?
esp> busy p... ...ignored

esp> ?
esp> busy p... ...ignored

esp> ?
esp> busy p... ...ignored

esp> ?
esp> busy p... ...ignored

esp> 4,CONNECT ...ignored
esp> OK ...matched
Connecting...
```

what i have tried
- connect to my hosting instead arduino.cc
- higher baudrate (115200, 250000)

but nothing changes

how can improve my sketch?
or is related to the library?

bialabs, 2019-10-03:
using an external antenna attached to the esp seems that works better!
I think is a RF interference problem because my esp is near the router
thank you for your advices

bialabs, 2019-10-04:
in which conditions the library responds with: To much garbage on RX ?

after that message client.connect stops working. How to avoid that?

jandrassy, 2019-10-04:
> in which conditions the library responds with: **To much garbage on RX** ?
> 
> after that error client.connect stops working. How to avoid that?

after 25 ignored lines waiting to match the expected response. usually there are ignored lines with some notifications from the AT firmware and at reset there are around 22 boot log lines. I implemented this to detect wrong connection for example with unconnected RX pin or without ground connected between Arduino and esp8266.

can you show the log?

bialabs, 2019-10-04:
i'll explain better my case
i am using your library in a project where i send the content of a multiple files (many) to a server.
Each file i make a client.connect and a get request.
The strange part is that i have this error **always** after 255 client.connect.
In the log i get:

```
esp> AT+CIPSTART=4,"TCP","www.myserver.com",80 ...sent

esp> ?
esp> busy p... ...ignored 
..(repeated 25 times)..

esp> ?
esp> busy p...esp ERROR: To much garbage on RX
```

for now i have found a dirty hack, when it happens i reinitialize the esp with:
```
Serial3.begin(9600);
WiFi.init(Serial3);
WiFi.begin(ssid, pass);
```

jandrassy, 2019-10-04:
it looks like a bug in AT firmware. if you want to try to wait longer, you can set longer timeout on Serial3 with `Serial3.setTimeout(2000);` (default is 1000 milliseconds)

bialabs, 2019-10-05:
setTimeout do not fix the issue with the at firmware

for now when it happens i reinitialize Wifi because in this situation i get IP Address 0.0.0.0 and 0 dBm signal

WerkAG, 2021-07-09:
I'm facing a very similar problem than described before. With the required changes, I ported a code which uses WiFiEsp library, to your WiFiEspAT library.
The code sends data to four diferent servers at diferent time intervals. Everything works well, but after some hours (usually between 12 to 18) client.connect stops working, and every attempt to connect will fail, even to a server on the local network.
When this happens, I can ping the device running the code, and even access the web server which serves me a little web page. So WiFi is working.
I don't think this problem is caused by an AT 1.7.4 firmware bug, because it not happens with the "old" WiFiEsp library, even running with AT 1.7.4  
Your library is a very interesting and usefull library, maybe you would like to check and / or try to reproduce the reported issue, because it really happens.
My next step will be add a counter to every client.connect event to try to verify, if the error happens after client.connect is called 255 times, as reported above by bialabs.
Thank you.
Rgs.