JAndrassy/WiFiEspAT #2: Access IP of stations connected to SoftAP

jandrassy, 2019-10-05:
wrap command AT+CWLIF

atesin, 2022-04-27:
is this issue still open?

i was thinking something elaborated...  but i realized this is already done in AdvancedChatServer example with WiFiServer.accept() managing our own external array of connected clients....  though i dont know what event to listen when a client leaves, other than polling :/

maybe using client.status() or query some other way ... we talked this before but i dont remember, there was a situation when client disconnected but there are still data unread on his buffer....  maybe everytime we know there is no more data we should also check if client is still connected to remove him from array if not

the another elaborated solution could be make the Server.atCwlif() return an iterable object that retuns a Client on each loop, until null.... or make it directly return an array of clients, or sockets (i am worried about memory comsumption though)
```
void loop()(
{
  WiFiClient client = WiFiServer.accept();
  if ( client )
    doSomethingWith(client);
}
```
```
void loop()(
{
  ClientsList clients = WiFiServer.atCwlif();
  while ( client = ClientList.next() )  // next, get, pull, pop, shift or whatever used in C/arduino
    doSomethingWith(client);
}
```
```
void loop()(
{
  WiFiClient clients[] = WiFiServer.atCwlif();
  int len = sizeof(clients) / sizeof(WiFiClient);
  for ( byte i = 0; i < len, ++i )
    doSomethingWith(clients[i]);
}
```

JAndrassy, 2022-04-27:
stations connected to SoftAP, not clients connected to a TCP server

atesin, 2022-04-27:
...  so ports and sockets and buffers are meaningless....  what about something like this?...
```
void loop()(
{
  IPAddress stations[WIFIESPAT_LINKS_COUNT];
  WiFi.atCwlif(stations);
  for ( byte i = 0; i < WIFIESPAT_LINKS_COUNT, ++i )
    doSomethingWith(stations[i]);
}
```

JAndrassy, 2022-04-27:
yes something like that.
I would name the function `softApStations`

atesin, 2022-04-27:
i was thinking a little.... i am looking the code, specially from `WiFi.scanNetworks()` to serve as a template for this new command... 

in documentation you say we can pass an optional parameter `WiFiApData[]` to fill the actual networks... and looking in the source it seems you should also pass the arrray length ... so what would happen in this situation?
```
WiFiApData networks[2];
WiFi.scanNetworks(networks); // with 8 networks found
```
sorry for replying here, maybe this is a matter for a new issue but as i found with occasion of implementing this new command... anyway i think you flaw badly with command documentation, in some time i will check the source and contribute with a more detailed api specification