JAndrassy/WiFiEspAT #3: "dynamic" print

atesin, 2019-11-05:
hi... thanks for your library, seems very cool .. i am trying to learn how to use, is a little hard to me since i am not a programmer but a sysadmin

i have an iot project (a cheap automated irrigation system for poor farmers) with arduino, this library, esp8266 (esp-01s) and the LoBo firmware (because the official firmware didn't let me save configs), and my nginx/php/mariadb webserver

i am developing a small embedded http server, mostly for inputting wifi credentials through a local webpage to connect the module and the server through internet... for example i know with this library i can do something like this...
```
// after receive +IPD and process http request url
void sendHtmlDashboard(WiFiClient client)
{
  client.flush(); // pseudocode to catch the idea
  char userName[20] = getUsernameSomeWay();
  char newMessages[10] = getNumMessagesSomeWay();

  client.print( F(
    "HTTP 200 OK\r\n"
    "Server: arduino/esp\r\n"
    // ... 2KB of additional http headers...
  );

  client.print( F(
    "<html>..."
    "<head, title, meta, style, script, etc... />"
    // ... 2KB of additional html headers...
  );

  client.print( F("welcome ") );
  client.print( userName );
  client.print( F(", you have ") );
  client.print( newMessages );
  client.print( F(" unread messages...") );
  // ... 2kb of additional dynamic webpage responses...

  client.stop();
}
```

... but i know with this will have several AT+CIPSEND commands, slowing down the overall responsiveness and resources.... so instead we can do it with 1 single cipsend command like this...
```
// after receive +IPD and process http request url
void sendHtmlDashboard(WiFiClient client)
{
  client.flush(); // pseudocode to catch the idea
  char userName[20] = getUsernameSomeWay();
  char newMessages[10] = getNumMessagesSomeWay();

  char response[4096] = '\0'; // note currently this can't be done with just 2kb ram

  strcpy_P( response, (PGM_P) F(
    "HTTP 200 OK\r\n"
    "Server: arduino/esp\r\n"
    // ... 2KB of additional http headers...
  );

  strcpy_P( response, (PGM_P) F(
    "<html>..."
    "<head, title, meta, style, script, etc... />"
    // ... 2KB of additional html headers...
  );

  strcpy_P( response, (PGM_P) F("welcome ") );
  strcpy( response, username );
  strcpy_P( response, (PGM_P) F(", you have ") );
  strcpy( response, messages );
  strcpy_P( response, (PGM_P) F(" new messages...") );
  // ... 2kb of additional webpage content in response...

  client.print( response ); // the single response
  client.stop();
}
```

... but with this way you must build the char string response before send it, which could consume a lot of ram.... i thought an alternative aproach that i don't know how to implement with this library...
```
// after receive +IPD and process http request url
void sendHtmlDashboard(WiFiClient client)
{
  clientl.flush(); // pseudocode to catch the idea
  char userName[20] = getUsernameSomeWay();
  char newMessages[10] = getNumMessagesSomeWay();

  // no more need to declare a char string with this techique, continue reading for details

  const __FlashStringHelper * httpHeaders = F(
    "HTTP 200 OK\r\n"
    "Server: arduino/esp\r\n"
    // ... 2KB of additional http headers...
  );

  const __FlashStringHelper * htmlHeaders = F(
    "<html>..."
    "<head, title, meta, style, script, etc... />"
    // ... 2KB of additional html headers...
  );

  const __FlashStringHelper * welcome = F("welcome ");
  const __FlashStringHelper * youHave = F(", you have ");
  const __FlashStringHelper * unreadMessages = F(" unread messages...") );
  // ... 2kb of additional webpage content in response...

  // send total data length in first place
  client.sendDataLength(
      strlen_P( (PGM_P) httpHeaders )
    + strlen_P( (PGM_P) htmlHeaders )
    + strlen_P( (PGM_P) welcome )
    + strlen( username )
    + strlen_P( (PGM_P) youHave )
    + strlen( newMessages )
    + strlen( (PGM_P) unreadMessages )
    // add additional content length if any
  );

  // ... then send the data itself, all at once
  // the esp8266 are already waiting "length" data because of the previous command
  client.sendData( httpHeaders );
  client.sendData( htmlHeaders );
  client.sendData( welcome );
  client.sendData( username );
  client.sendData( youHave )
  client.sendData( newmessages );
  client.sendData( unreadMessages );
  // send additional webpage content if any

  // done, now just close the link
  client.stop();
}
```

... now the questions...

- can something like this currently be done with this library?
- HOW can something like this currently be achieved with this library?
- if not, when something like this will be implemented in this library?

thanks for your fantastic library, for your patience reading, and in advance for your possible response...  i am hungry waiting your response :D

jandrassy, 2019-11-06:
The WiFiEspAT library[ uses a buffer](https://github.com/jandrassy/WiFiEspAT#buffering-and-buffers) so the prints are collected and sent as 32 or 64 bytes pieces. You can configure the buffer size.

I have a separate [StreamLib](https://github.com/jandrassy/StreamLib) library as solution for things you mention. My main Arduino project Regulator is the largest example of StreamLib with web server and client.

atesin, 2019-11-06:
hi...  thanks for your kindly response ...  but how about the 3 questions?

i am not so skilled... i am cracking my head trying to find some custom "AT+CIPSEND:<link>,<numBytes> ; esp.find('>') ; esp.print(...) ; esp.print(...) ; esp.print(...) ..." with this library :) ....  

another additional question... is there some WiFiClient.printRaw() or WiFiClient.find() function?

jandrassy, 2019-11-07:
I answered the questions with links, but OK, I go a little more in detail, but you must read the docs and study the code

WFiClient is derived from Arduino common Print class so you can use all functions you know from Serial, including printing a F() macro string. 
With StreamLib classes you have additionally [printf](http://www.cplusplus.com/reference/cstdio/printf/), which can use format string from PROGMEM too

the[ HTTP chunked encoding](https://en.wikipedia.org/wiki/Chunked_transfer_encoding) with ChunkedPrint class solves the unknown-http-body-length problem as you can read in StreamLib's README.md

my code as examples (it doesn't use WiFiEspAT, but the networking clients are all the same)

- [POST request 1](https://github.com/jandrassy/Regulator/blob/4c1f15e00f51d618293ae441706c3f6027a56405/Regulator/WemoInsight.ino#L47)
- [POST request 2](https://github.com/jandrassy/Regulator/blob/4c1f15e00f51d618293ae441706c3f6027a56405/Regulator/SusCalib.ino#L50)
- [web server client](https://github.com/jandrassy/Regulator/blob/4c1f15e00f51d618293ae441706c3f6027a56405/Regulator/WebServer.ino#L22)

closed 2019-11-07 by jandrassy