JAndrassy/WiFiEspAT #4: Ip address error in Wifi Client mode

sgamoct, 2019-11-11:
Hi,
my esp8266, with At 1.7 command release,it's set in client mode (temporary o super persistent connection) received ip address isn't correct.
My WiFi network dchp server have ip range 192.168.0.2 to 255 but ip recived in serial monitor from esp8266 is 65.38.83.95. This received address will not ping from cmd. Work only whith fixed ip address, gatway and subnetmask.
can you help me?

jandrassy, 2019-11-12:
please change in src/utils/EspAtDrvLogging.h to `#define LOG_LEVEL LOG_LEVEL_DEBUG`
to see the AT commands

sgamoct, 2019-11-12:
hi i didn't understand the change  because My file in src/utils/EspAtDrvLogging.h it's 

#define LOG_LEVEL_SILENT 0
#define LOG_LEVEL_ERROR  1
#define LOG_LEVEL_WARN   2
#define LOG_LEVEL_INFO   3
#define LOG_LEVEL_DEBUG  4

I send the AT commands manually from  console and i have the same ip problem. 
any ideas?

jandrassy, 2019-11-12:
there is #define LOG_LEVEL LOG_LEVEL_SILENT in EspAtDrvLogging.h

you could try AT 1.7.1 version

sgamoct, 2019-11-12:
Unfortunately doesn't work.
I can use ESP only fixed ip address in this moment, but is'nt a good solution.
I tried AT 1.7.1 version  with same problems.
I had upgrade ESP from 1.3 sdk2 512+512 to 1.7 sdk 3 1024+1024 using 16Mb-c1 option.  I'm not sure but Problems will depending from upgrade? now i will try to downgrade at AT1.62. this version it's unsupported in WiFIEspAT?

jandrassy, 2019-11-12:
only AT 1.7 and 1.7.1 have the 'passive mode' used by the library. older versions and AT 2.0 don't have 'passive mode'.
Espressif works on 'passive mode' for 2.1. We will see whether they will succeed. Right now in git version and ESP32 AT 2.0 release UDP doesn't work at all in passive mode.

sgamoct, 2019-11-12:
My ESP8266 is a model with only 512 + 512 user memories.
I downgraded the user.bin option 8 Mb and 512 + 512 now from 1.7 to 1.62. In this case the IP address in dhcp mode works.
Now I'm sure the problem is upgrading to the 1.7 AT file size. I ran the update with 1024 + 1024 user.bin and selected the 16 MB -c1 option. so, excluse the dynamic ip, with this solution everything seems to work.

jandrassy, 2019-11-12:
use 2MB-c1 0x0 boot_v1.7.bin 0x01000 at/1024+1024/user1.2048.new.5.bin 0x1fb000 blank.bin 0x1fc000 esp_init_data_default_v08.bin 0xfe000 blank.bin 0x1fe000 blank.bin
on any flash size except 512kB

sgamoct, 2019-11-12:
This is the flash set up. correct?
[image]

after flash i run in terminal At command and this it's the result:
[image]
but my network gatway it' 192.168.0.1.

jandrassy, 2019-11-12:
try the AT Lobo build for 1 MB flash
https://github.com/loboris/ESP8266_AT_LoBo/#flashing

sgamoct, 2019-11-13:
i have try 1.7 and 1.71 espressif AT without success.
For LoBo AT i can compile only in linux machine? i 'm not linux frienly user  and not have competente for compiling correct file :(.
thx for all.

jandrassy, 2019-11-13:
> 
> 
> i have try 1.7 and 1.71 espressif AT without success.
> For LoBo AT i can compile only in linux machine? i 'm not linux frienly user and not have competente for compiling correct file :(.
> thx for all.

there are compiled binaries. install esptool and the flash.sh script should run in Windows Power Shell

sgamoct, 2019-11-13:
please can you link the correct file i will download?

jandrassy, 2019-11-13:
bin/upgrade/esp8266_AT_1}_2.bin should be the right for 1 MB flash
the other bin files should go to addresses as for the normal AT firmware for 1MB flash

sgamoct, 2019-11-13:
Thank you, i try it now. replace \user1.1024.new.2.bin with esp8266_AT_1}_2.bin file in the same memory address?

jandrassy, 2019-11-13:
> Thank you, i try it now. replace \user1.1024.new.2.bin with esp8266_AT_1_2.bin file in the same memory address?

yes 0x1000

sgamoct, 2019-11-13:
> > Thank you, i try it now. replace \user1.1024.new.2.bin with esp8266_AT_1_2.bin file in the same memory address?
> 
> yes 0x1000

dosen't work. i now write in the github LoBo for ask the correct configuration. buil.sh not work in my windows power shell.
Thankyou for all i will try to resolve ...

jandrassy, 2019-11-13:
how it doesn't work?

sgamoct, 2019-11-13:
> how it doesn't work?

I received error in terminal after flash:

No configured firmware found, check all...
Address: 001000
Wrong flash mode
Address: 101000
Wrong flash mode
FATAL ERROR, no firmware found

[image]

[image]

jandrassy, 2019-11-13:
the addresses are not right for 1 MB
blank.bin 0xFB000
esp_init_data_default.bin 0xFC000 
blank.bin 0x7E000 
blank.bin 0xFE000

sgamoct, 2019-11-13:
Happy Happy!
Very thanks my friend. All work fine ESP fimware and your WiFiESP library. Thank you very much for you support.

closed 2019-11-13 by jandrassy