JAndrassy/WiFiEspAT #6: How to print esp client hostname?

heinemannj, 2020-01-13:
I'm using a persistent WiFi connection with dhcp enabled in client mode.
How to print esp client hostname?

Goal:
I want to feed an mqtt_clientId with the default esp hostname

const char* mqtt_clientId = "WEMOSMega_A0FFFF";

jandrassy, 2020-01-13:
Sorry, right now there isn't a function to read the hostname. I will add it tomorrow.

jandrassy, 2020-01-14:
can you try the 'git version'?

heinemannj, 2020-01-16:
Hi Juraj,
I confirm that WiFi.hostname() in the new measter version is working for me:

  char hostname[33];
  WiFi.hostname(hostname);
  Serial.print("hostname ");
  Serial.println(hostname);

11:42:41.563 -> hostname ESP-C60650

Many thanks for your support!
Cheers