JAndrassy/WiFiEspAT #10: Current UART Configuration - AT+UART_CUR?

heinemannj, 2020-02-06:
I wants to automate my AT_BAUD_RATE configuration to ensure that the sketch is using the same baud rate as the actual configuration on ESP.

// Baud rate for communication with AT firmware
#define AT_BAUD_RATE 500000           // For Arduino Mega over Hardware Serial "500000" baud works good

Can you please implement?

jandrassy, 2020-02-06:
without the common baud rate the devices can't negotiate the baud rate.

heinemannj, 2020-02-06:
Comments in your ChangeATBaudRate.ino:

... but is able to send a short string at 115200 baud. So we send at 115200 baud ... the AT command to change the baud rate to 9600 baud ...

Maybe it is also possible to use the default 115200 baud rate to identify the current UART Configuration?

jandrassy, 2020-02-06:
the comment is about SoftwareSerial not about AT firmware. we can send over SoftwareSerial at 115200 baud if we know that he firmware listens at this rate. but SoftwareSerial can't reliably listen at 115200 baud.

closed 2020-02-08 by jandrassy