JAndrassy/WiFiEspAT #11: Ignored AT commands during WiFi setup

heinemannj, 2020-02-06:
AT version:1.7.1.0(Jul 15 2019 16:58:04)
SDK version:3.0.1(78a3e33)
compile time:Jul 15 2019 18:12:41

My WiFi configuration:

//=======================================================================
//                             Setup Wifi
//=======================================================================
IPAddress wifiIP;
uint8_t wifiMac[6];
char wifiMacStr[18];
char wifiHostname[11];
long wifiRssi;

unsigned long wifiErrorCounter = 0;

void setup_wifi() {
  // Initialize Serial port for communication with the WiFi module
  SerialAT.begin(AT_BAUD_RATE);
  WiFi.init(SerialAT);

  // Check if Serial port for communication with the WiFi module is working
  if (WiFi.status() == WL_NO_MODULE) {
    Serial.println();
    Serial.println("Communication with WiFi module failed!");
    // don't continue
    while (true) {
      blink(100);
    }
  }

  // Disable ESP deep sleep mode
  Serial.println("Sending AT deep sleep change (disable deep sleep) ...");
  SerialAT.println("AT+SLEEP=0");
  blink(500);

  // Disable default automatic start of persistent AP
  WiFi.endAP(true);

  // Waiting for connection to Wifi network set with the SetupWiFiConnection sketch
  Serial.println("Waiting for connection to WiFi");
  while (WiFi.status() != WL_CONNECTED) {
     blink(500);
    Serial.print('.');
  }
  Serial.println();
  Serial.println("Connected to WiFi network.");

  // SSID of the network you're attached to:
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  // Your WiFi shield's IP address:
  wifiIP = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(wifiIP);

  // Your WiFi shield's MAC address
  WiFi.macAddress(wifiMac);
  snprintf(wifiMacStr, sizeof(wifiMacStr), "%02X:%02X:%02X:%02X:%02X:%02X", wifiMac[5], wifiMac[4], wifiMac[3], wifiMac[2], wifiMac[1], wifiMac[0]);
  Serial.print("MAC Address: ");
  Serial.println(wifiMacStr);

  // Your WiFi shield's hostname:
  snprintf(wifiHostname, sizeof(wifiHostname), "ESP-%02X%02X%02X", wifiMac[2], wifiMac[1], wifiMac[0]);
  Serial.print("Hostname: ");
  Serial.println(wifiHostname);

  // The received WiFi signal strength:
  wifiRssi = WiFi.RSSI();
  Serial.print("Signal Strength (RSSI): ");
  Serial.print(wifiRssi);
  Serial.println(" dBm");
}

16:27:03.665 -> esp INFO: soft reset
16:27:03.665 -> esp> AT+RST ...sent
16:27:03.665 -> esp>  ...ignored
16:27:03.665 -> esp> ⸮ ...ignored
16:27:03.665 -> esp>  ...ignored
16:27:03.698 -> esp> ⸮ ...ignored
16:27:03.698 -> esp> ⸮ ...ignored
16:27:03.766 -> esp>  ...ignored
16:27:03.800 -> esp> ⸮⸮⸮⸮⸮ ...ignored
16:27:03.800 -> esp>  ...ignored
16:27:03.800 -> esp> ⸮ ...ignored
16:27:03.800 -> esp>  ...ignored
16:27:03.800 -> esp>  ...ignored
16:27:03.800 -> esp> ⸮ ...ignored
16:27:03.800 -> esp> ⸮ ...ignored
16:27:03.800 -> esp> ⸮⸮⸮⸮ ...ignored
16:27:03.800 -> esp> ready ...matched
16:27:03.800 -> esp> ATE0 ...sent
16:27:03.800 -> esp> ATE0 ...ignored
16:27:03.800 -> esp> OK ...matched
16:27:03.800 -> esp> AT+CIPMUX=1 ...sent
16:27:03.800 -> esp> OK ...matched
16:27:03.800 -> esp> AT+CIPRECVMODE=1 ...sent
16:27:03.834 -> esp> OK ...matched
16:27:03.834 -> esp> AT+CWMODE? ...sent
16:27:03.834 -> esp> +CWMODE:1 ...matched
16:27:03.834 -> esp> OK ...matched
16:27:03.834 -> esp INFO: wifi status
16:27:03.834 -> esp> AT+CIPSTATUS ...sent
16:27:03.834 -> esp> STATUS:5 ...matched
16:27:03.834 -> esp> OK ...matched
16:27:03.834 -> Sending AT deep sleep change (disable deep sleep) ...
16:27:04.833 -> esp> OK ...ignored
16:27:04.833 -> esp> WIFI CONNECTED ...ignored
16:27:04.833 -> esp> WIFI GOT IP ...ignored
16:27:04.833 -> esp INFO: end SoftAP persistent
16:27:04.833 -> Waiting for connection to WiFi
16:27:04.833 -> esp INFO: wifi status
16:27:04.833 -> esp> AT+CIPSTATUS ...sent
16:27:04.867 -> esp> STATUS:2 ...matched
16:27:04.867 -> esp> OK ...matched
16:27:04.867 -> 
16:27:04.867 -> Connected to WiFi network.
16:27:04.867 -> SSID: esp INFO: AP query
16:27:04.867 -> esp> AT+CWJAP? ...sent
16:27:04.867 -> esp> +CWJAP:"area053","e0:28:6d:cf:f1:e1",11,-73,0 ...matched
16:27:04.867 -> esp> OK ...matched
16:27:04.867 -> area053
16:27:04.867 -> esp INFO: STA IP query
16:27:04.867 -> esp> AT+CIPSTA? ...sent
16:27:04.867 -> esp> +CIPSTA:ip:"192.168.178.35" ...matched
16:27:04.867 -> esp> +CIPSTA:gateway:"192.168.178.1" ...matched
16:27:05.883 -> esp> +CIPSTA:netmask:"255.255.255.0 ...matched
16:27:06.864 -> esp> ?
16:27:06.864 -> esp> ERROR ...error
16:27:06.864 -> esp ERROR: expected OK got ERROR
16:27:06.864 -> IP Address: 192.168.178.35
16:27:06.898 -> esp INFO: STA MAC query 
16:27:06.898 -> esp> AT+CIPSTAMAC? ...sent
16:27:06.898 -> esp> +CIPSTAMAC:"a4:cf:12:c6:06:50" ...matched
16:27:06.898 -> esp> OK ...matched
16:27:06.898 -> MAC Address: A4:CF:12:C6:06:50
16:27:06.898 -> Hostname: ESP-C60650
16:27:06.898 -> esp INFO: AP query
16:27:06.898 -> esp> AT+CWJAP? ...sent
16:27:06.898 -> esp> +CWJAP:"area053","e0:28:6d:cf:f1:e1",11,-72,0 ...matched
16:27:06.898 -> esp> OK ...matched
16:27:06.898 -> Signal Strength (RSSI): -72 dBm

jandrassy, 2020-02-06:
what commands? only the bootlog and information sent by AT firmware are ignored

heinemannj, 2020-02-06:
OK - If you confirm that I'm not running into issues I feel much more relaxed ...

What's about the following?

16:27:03.665 -> esp INFO: soft reset
16:27:03.665 -> esp> AT+RST ...sent
16:27:03.665 -> esp> ...ignored

jandrassy, 2020-02-06:
only the INFO line would be logged at INFO log level. the rest is logged only at DEBUG level.
AT+RST is the reset command. then the start up of esp8266 has some boot log at different baud rate so we receive garbage. the library waits for "ready" to continue

heinemannj, 2020-02-07:
Conclusion:
The garbage in the debug is NO technical issues  - Misinterpretation from my side

Thanks for your helpful expanations!

closed 2020-02-07 by heinemannj