JAndrassy/WiFiEspAT #9: No stable tcp connections - Disable ESP deep sleep mode

heinemannj, 2020-02-06:
I was running into a nightmare - no stable tcp connections to my MQTT server ... 

The following has fixed my painful issue:

 // Disable ESP deep sleep mode
  Serial.println("Sending AT deep sleep change (disable deep sleep) ...");
  SerialAT.println("AT+SLEEP=0");

Can you please document and implement?

jandrassy, 2020-02-06:
what version of AT firmware did you use?
I could add a WiFi.sleepMode(mode) function

heinemannj, 2020-02-06:
AT version:1.7.1.0(Jul 15 2019 16:58:04)
SDK version:3.0.1(78a3e33)
compile time:Jul 15 2019 18:12:41

But there is already a new update available: 1.7.2
If you want I can update to 1.7.2

PLEASE update also your documentation - this issue makes really NO fun ...

jandrassy, 2020-02-06:
I didn't know about 1.7.2.

jandrassy, 2020-02-13:
new commit. can you try the 'git' version?
`sleepMode`- to set the level of automatic sleep mode. possible modes are WIFI_NONE_SLEEP, WIFI_LIGHT_SLEEP and WIFI_MODEM_SLEEP

heinemannj, 2020-02-16:
Thank you for the new feature!

I confirm that disabling of ESP sleep mode is working as expected:

14:09:00.251 -> ##### Starting Setup ######
14:09:00.251 -> 
14:09:00.251 -> Version: 0.0.58
14:09:00.251 -> Setting AT baud rate
14:09:01.265 -> Using Hardware Serial
14:09:01.265 -> esp> ⸮⸮ ...ignored
14:09:01.265 -> esp INFO: soft reset
14:09:01.265 -> esp> AT+RST ...sent
14:09:01.265 -> esp> AT+RST ...ignored
14:09:01.265 -> esp> OK ...ignored
14:09:01.265 -> esp> WIFI DISCONNECT ...ignored
14:09:01.366 -> esp> ⸮ ...ignored
14:09:01.366 -> esp>  ...ignored
14:09:01.366 -> esp>  ...ignored
14:09:01.366 -> esp> ⸮⸮ ...ignored
14:09:01.400 -> esp> ⸮⸮⸮ ...ignored
14:09:01.400 -> esp> ⸮ ...ignored
14:09:01.400 -> esp>  ...ignored
14:09:01.400 -> esp> ⸮ ...ignored
14:09:01.400 -> esp>  ...ignored
14:09:01.400 -> esp> ⸮ ...ignored
14:09:01.400 -> esp> ⸮ ...ignored
14:09:01.400 -> esp> ⸮⸮⸮ ...ignored
14:09:01.433 -> esp> ⸮⸮⸮⸮⸮⸮ ...ignored
14:09:01.501 -> esp>  ...ignored
14:09:01.501 -> esp> ⸮⸮⸮ ...ignored
14:09:01.501 -> esp> ⸮⸮⸮ ...ignored
14:09:01.501 -> esp> ⸮⸮⸮ ...ignored
14:09:01.501 -> esp>  ...ignored
14:09:01.501 -> esp>  ...ignored
14:09:01.501 -> esp> ⸮ ...ignored
14:09:01.501 -> esp> ⸮⸮⸮⸮ ...ignored
14:09:01.501 -> esp>  ...ignored
14:09:01.501 -> esp> ready ...matched
14:09:01.501 -> esp> ATE0 ...sent
14:09:01.501 -> esp> ATE0 ...ignored
14:09:01.501 -> esp> OK ...matched
14:09:01.501 -> esp> AT+CIPMUX=1 ...sent
14:09:01.501 -> esp> OK ...matched
14:09:01.535 -> esp> AT+CIPRECVMODE=1 ...sent
14:09:01.535 -> esp> OK ...matched
14:09:01.535 -> esp> AT+CWMODE? ...sent
14:09:01.535 -> esp> +CWMODE:1 ...matched
14:09:01.535 -> esp> OK ...matched
14:09:01.535 -> Disable WiFi module deep sleep
14:09:01.535 -> esp INFO: set sleep mode
14:09:01.535 -> esp> AT+SLEEP=0 ...sent
14:09:01.535 -> esp> OK ...matched
14:09:01.535 -> esp INFO: wifi status
14:09:01.535 -> esp> AT+CIPSTATUS ...sent
14:09:01.535 -> esp> STATUS:5 ...matched
14:09:01.535 -> esp> OK ...matched
14:09:01.535 -> esp INFO: fw version
14:09:01.535 -> esp> AT+GMR ...sent
14:09:01.535 -> esp> AT version:1.7.2.0(Jan 19 2020 09:39:04) ...matched
14:09:01.569 -> esp> SDK version:3.0.2(824dc80) ...ignored
14:09:01.569 -> esp> compile time:Jan 20 2020 10:09:38 ...ignored
14:09:01.772 -> esp> WIFI CONNECTED ...ignored
14:09:02.449 -> esp> WIFI GOT IP ...ignored
14:09:03.461 -> esp> ?
14:09:03.461 -> esp> ERROR ...error
14:09:03.461 -> esp ERROR: expected OK got ERROR
14:09:03.461 -> AT baud rate: 500000
14:09:03.461 -> AT data bits: 8 (8-bit data)
14:09:03.461 -> AT stop bits: 1 (1-bit stop bit)
14:09:03.461 -> AT parity bit: 0 (None)
14:09:03.461 -> AT flow control: 0 (flow control is not enabled)
14:09:03.461 -> AT firmware version: 1.7.2.0
14:09:03.461 -> AT firmware is OK for the WiFiEspAT library
14:09:03.494 -> esp INFO: end SoftAP persistent
14:09:03.494 -> AP failed to start or is disabled
14:09:03.494 -> Waiting for connection to WiFi network esp INFO: wifi status
14:09:03.494 -> esp> AT+CIPSTATUS ...sent
14:09:03.494 -> esp> STATUS:2 ...matched
14:09:03.494 -> esp> OK ...matched
14:09:03.494 -> 
14:09:03.494 -> Connected to WiFi network [persistent]

This fix improve the ESP stability but is not resolving my problems at all ...
I will open new tickets.

Thanks and cheers

closed 2020-02-16 by jandrassy