JAndrassy/WiFiEspAT #12: Modem Sleep function

sgamoct, 2020-02-13:
Hi,
i can use a modem sleep function with your library?
how is the command? i can't use deep sleep with esp01 beacuse don't have a wakeup reset pin

jandrassy, 2020-02-13:
it is the next thing I want to add to library. maybe today or tomorrow

sgamoct, 2020-02-13:
Thx i waiting you

jandrassy, 2020-02-13:
from reference

> AT+SLEEP=mode
> Notes
> ‣ 0: disables sleep mode
> ‣ 1: Light-sleep mode
> ‣ 2: Modem-sleep mode
> This command can only be used in Station mode. 
> **Modem-sleep is the default sleep mode.**

jandrassy, 2020-02-13:
why you can't wire reset pin for wake-up from deep sleep? esp-01 has the reset pin on the 4x2 header

sgamoct, 2020-02-13:
> his command can only be used in Station mode.

don't work in client mode?

sgamoct, 2020-02-13:
> perché non riesci a cablare il pin di ripristino per il risveglio dal sonno profondo? esp-01 ha il pin di reset sull'intestazione 4x2

I read, to enable sleep mode in deep sleep, it must be added to new wire.See the image
[image]

jandrassy, 2020-02-13:
io 16 to reset is for timed deep sleep where the esp8266 wakes up itself. we use endless deep sleep and wake-up by Arduino. see the DeepSleepAndHwReset example.

jandrassy, 2020-02-13:
> > his command can only be used in Station mode.
> 
> don't work in client mode?

STA is station on WiFi network. some call it client, but it is confusing about the roles of server and client on TCP level. STA can run a TCP server to which a TCP client running on AP  connects.
AP manages the physical (RF) part of the network.

sgamoct, 2020-02-13:
> io 16 to reset is for timed deep sleep where the esp8266 wakes up itself. we use endless deep sleep and wake-up by Arduino. see the DeepSleepAndHwReset example.

Ok i try It Tomorrow

sgamoct, 2020-02-13:
> STA is station on WiFi network. some call it client, but it is confusing about the roles of server and client on TCP level. STA can run a TCP server to which a TCP client running on AP connects.
> AP [...]

THX for the explanation my friend

closed 2020-02-16 by jandrassy

sgamoct, 2020-02-18:
Hi i have just try, all work fine.
for deepsleep i have used AT+GSLP=xxx (time in ms).only a question: Why during X time the quiescent current it's 0.500mA but after the deepsleep time the current it's 11.7 mA but esp it's correctly in deepsleep?for exit to deepsleep put Reset to gnd.
 very thanks!

sgamoct, 2020-02-26:
Hi i have try to insert the command in my program but now not working.
i have use this command for sleep1:
 void ReteOff() {
    Serial1.print ("AT+SLEEP=1");
}
for sleep0:
 void ReteOn() {
    Serial1.print ("AT+SLEEP=0");
}

why?
thx

jandrassy, 2020-02-27:
WiFi.sleepMode()
to set the level of automatic sleep mode. possible modes are WIFI_NONE_SLEEP, WIFI_LIGHT_SLEEP and WIFI_MODEM_SLEEP

sgamoct, 2020-02-27:
Thankyou, i have download the new version 
it's this The correct syntax 
WiFi.sleepMode(WIFI_NONE_SLEEP);
WiFi.sleepMode(WIFI_MODEM_SLEEP);
?