JAndrassy/WiFiEspAT #16: Custom AT Command Processor with TLS 1.2

JiriBilek, 2020-05-20:
Hi Juraj,
I am testing your library with my custom AT command processor (https://github.com/JiriBilek/ESP_ATMod).
It's quite a new project still in WIP status but I tested your library with SSL connection and it runs.

The reasons I made the ESP_ATMod project are:
- lack of state-of-the-art TLS ciphersuites in Espressif AT firmware and all its derivations (e.g. LoBo);
- no checking of TLS certificate at all - a big security hole. I implemented fingerprint and/or certificate checking commands. I think that we should not use insecure connection where possible.
- the big binary of the current AT firmware can't be used on ESP-01. And this tiny module is the one I think is the most suitable for AT firmware (quite limited GPIOs prevent to use it in native applications).

I implemented only station mode and only TCP/SSL mode. No AP nor UDP connections.

I'd be very glad if you tried the firmware. If you want to make a SSL connection, just substitute "TCP" keyword with "SSL".

jandrassy, 2020-05-21:
Hi Jiri. Thank you, I try it. And I agree with you that the esp-01 modules as preferred for AT firmware. Even better the esp-01S with the simple wiring. 

btw. Espressif works on SSL passive mode in the RTOS AT firmware 2.0 for esp32 and esp8266. But that firmware is large and buggy. I have an unpublished version of this library for AT 2.0 and SSL works on esp32 but has problems on esp8266. And UDP passive mode is still unfinished.

jandrassy, 2020-05-21:
works perfect. my test for which the AT 2.0 on esp8266 fails because it is a large SSL download, run without problems with your firmware.
I commited SSL support to library.

Jiri, prosim nepovedz ludom od esp8266 Arduino, ze si napisal AT firmware nad ich core. neviem ako by to zvladli :-)

closed 2020-05-21 by jandrassy

JiriBilek, 2020-05-21:
Ahoj, nebudu jim to tam psát, ledaže bych našel někde chybu, ale to je krajně nepravděpodobné.

Thanks for SSL support, I'll test it ASAP. 
It was TLS that started the development of my ESP_ATMod, because all my IOT devices are connected to active24.cz hosting and they removed all weak ciphers from TLS server ciperhsuite list and suddenly, all my devices stopped working :(