JAndrassy/WiFiEspAT #20: AT firmware not responding - MQTT TCP

dannykio, 2020-09-05:
Hi Juraj,

For starter... ESP8266 firmware 👍 
```
20:40:40.958 -> AT+GMR
20:40:40.958 -> AT version:1.7.4.0(May 11 2020 19:13:04)
20:40:40.992 -> SDK version:3.0.4(9532ceb)
20:40:41.025 -> compile time:May 27 2020 10:12:17
20:40:41.059 -> Bin version(Wroom 02):1.7.4
20:40:41.092 -> OK
```

PLATFORM: Arduino Uno/Mega + ESP8266 (UART)

ESP connects to the WiFi network correctly:
```
20:32:12.297 -> esp> AT+CIPMUX=1 ...sent
20:32:12.414 -> esp> OK ...matched
20:32:14.513 -> esp> AT+CIPSTATUS ...sent
20:32:14.513 -> esp> STATUS:5 ...matched
20:32:14.513 -> esp> OK ...matched
20:32:15.525 -> esp> WIFI GOT IP ...ignored
20:32:15.525 -> esp INFO: wifi status
20:32:15.525 -> esp> AT+CIPSTATUS ...sent
20:32:15.562 -> esp> STATUS:2 ...matched
20:32:15.562 -> esp> OK ...matched
20:32:15.562 -> 
20:32:15.562 -> Connected to WiFi
20:32:15.562 -> 
20:32:15.562 -> IPv4 Address: esp INFO: STA IP query
20:32:15.612 -> esp> AT+CIPSTA? ...sent
20:32:15.612 -> esp> +CIPSTA:ip:"192.168.1.229" ...matched
20:32:15.645 -> esp> +CIPSTA:gateway:"192.168.1.1" ...matched
20:32:15.680 -> esp> +CIPSTA:netmask:"255.255.255.0" ...matched
```

**PROBLEM:**
```
20:32:17.772 -> esp INFO: free linkId is 4
20:32:17.772 -> esp INFO: start TCP to hairdresser.cloudmqtt.com:16068 on link 4
20:32:17.812 -> esp> AT+CIPSTART=4,"TCP","hairdresser.cloudmqtt.com",16068 ...sent
20:32:18.840 -> esp> ?
20:32:19.839 -> esp> ?
20:32:20.858 -> esp> ?
20:32:21.845 -> esp ERROR: AT firmware not responding
20:32:21.895 -> 
20:32:21.895 -> MQTT: -2
20:32:24.976 -> esp>  ...ignored
```

I can't connect to the MQTT cloud broker... I have no idea why it's happening because I have tested it few times without your library - just Arduino console. Here's the example of what I did:

```
20:39:10.139 -> AT
20:39:10.139 -> 
20:39:10.139 -> OK

// AT+CIPMUX=0

20:39:28.255 -> AT+CIPSTART=4,"TCP","hairdresser.cloudmqtt.com",16068
20:39:28.304 -> Link type ERROR
20:39:28.304 -> 
20:39:28.304 -> ERROR

// AT+CIPMUX=1

20:39:44.003 -> AT+CIPMUX=1
20:39:44.003 -> 
20:39:44.003 -> OK
20:39:54.825 -> AT+CIPSTART=4,"TCP","hairdresser.cloudmqtt.com",16068
20:39:54.908 -> 4,CONNECT
20:39:54.908 -> 
20:39:54.908 -> OK
```

However I saw that WiFiEspAT sets up the AT+CIPMUX=1 in the beginning.

```
20:32:12.297 -> esp> AT+CIPMUX=1 ...sent
20:32:12.414 -> esp> OK ...matched
```

What can cause such a stupid... bug? **I use PubSubClient library by knolleary.**

I would be very thankful for your help. :)

~ Daniel

dannykio, 2020-09-06:
There's also a strange thing happening:
```
11:59:18.130 -> IPv4: 0.0.0.0
11:59:28.229 -> MQTT: -2
```
**It showed IP on beginning and now there's 0?**

jandrassy, 2020-09-06:
Hi, this works
```
#include <WiFiEspAT.h>
#include <PubSubClient.h>

const char* server = "hairdresser.cloudmqtt.com";

void callback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i=0;i<length;i++) {
    Serial.print((char)payload[i]);
  }
  Serial.println();
}

WiFiClient ethClient;
PubSubClient client(ethClient);

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    // Attempt to connect
    if (client.connect("arduinoClient")) {
      Serial.println("connected");
      // Once connected, publish an announcement...
      client.publish("outTopic","hello world");
      // ... and resubscribe
      client.subscribe("inTopic");
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}

void setup()
{
  Serial.begin(115200);

  client.setServer(server, 16068);
  client.setCallback(callback);

  Serial1.begin(115200);
  WiFi.init(Serial1);

  if (WiFi.status() == WL_NO_MODULE) {
    Serial.println();
    Serial.println("Communication with WiFi module failed!");
    // don't continue
    while (true);
  }

  // waiting for connection to Wifi network set with the SetupWiFiConnection sketch
  Serial.println("Waiting for connection to WiFi");
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.print('.');
  }
  Serial.println();
  Serial.println("Connected to WiFi network.");
}

void loop()
{
  if (!client.connected()) {
    reconnect();
  }
  client.loop();
}
```

do you do something with Serial used for esp8266? do you have interrupts disabled? do you call WiFiEspAT from interrupt?

dannykio, 2020-09-06:
Hi, thank you for your response. I can confirm that ESP connects to MQTT using the code above.

`do you do something with Serial used for esp8266? `

> Serial for ESP is separated from Serial used for logs. Both work on different baud rate.

To make this easier to understand I will send you the Sketch. ;)

dannykio, 2020-09-07:
Hi Juraj,

you were right... I didn't call ESP from interrupt. Adding the `sei()` fixed all problems.

Everything works perfectly - thank you for creating such a useful library for ESP8266! :)

~ Daniel

closed 2020-09-07 by dannykio