JAndrassy/WiFiEspAT #21: Google Firebase support

sgamoct, 2020-09-10:
Hi, can you add firebase support or a sample sketch for use it?
thankyou

jandrassy, 2020-09-10:
try the library Firebase for WiNINA

sgamoct, 2020-09-10:
Hi,
 i have try it but recive this error
Error compiling for the Arduino Mega or Mega 2560 board.
I have arduino mega with esp-01 connected via serial

jandrassy, 2020-09-11:
> Hi,
> i have try it but recive this error
> Error compiling for the Arduino Mega or Mega 2560 board.
> I have arduino mega with esp-01 connected via serial

Search for the error message in the console window. "Error compiling" is only an information that there was one or more errors.

sgamoct, 2020-09-11:
> > Hi,
> > i have try it but recive this error
> > Error compiling for the Arduino Mega or Mega 2560 board.
> > I have arduino mega with esp-01 connected via serial
> 
> Search for the error message in the console window. "Error compiling" is only an information that there was one or more errors.
> 
> in your sketch add
> #include <WiFiEspAT.h>
> before
> #include "Firebase_Arduino_WiFiNINA.h"
> 
> in Firebase_Arduino_WiFiNINA_HTTPClient.h change
> #include <SPI.h>
> #include <WiFiNINA.h>
> to
> #include <WiFi.h>
> 
> it will compile

Hi, thankyou for your support, I have try in this mode:
  ```
#include<LiquidCrystal.h>       
  #include <WiFiEspAT.h>          
  #include <WiFiClient.h>          
  #include "Firebase_Arduino_WiFiNINA.h"

```
and i have modify file Firebase_Arduino_WiFiNINA_HTTPClient.h 

```
#ifndef Firebase_Arduino_WiFiNINA_HTTPClient_H
#define Firebase_Arduino_WiFiNINA_HTTPClient_H

#include <Arduino.h>
#include <WiFi.h>
```
but not compile i have 2 error 
```
.../src/Firebase_Arduino_WiFiNINA_HTTPClient.h:100:5: error: 'WiFiSSLClient' does not name a type; did you mean 'WiFiClient'?

     WiFiSSLClient client;

     ^~~~~~~~~~~~~

     WiFiClient
```
and
```
src/Firebase_Arduino_WiFiNINA.h:736:5: error: 'WiFiSSLClient' does not name a type; did you mean 'WiFiClient'?

     WiFiSSLClient getWiFiClient();

     ^~~~~~~~~~~~~

     WiFiClient
```

jandrassy, 2020-09-11:
you have some old version of the WiFiEspAT library?

and then for SSL you must use AT 2.0.1

sgamoct, 2020-09-11:
👍 perfect, i have an older version. now i don't have error. now i write sketch for try firebase.
thank you for you support.

closed 2020-10-23 by jandrassy