JAndrassy/WiFiEspAT #23: DNS fail in WebClient example

charizord, 2020-10-03:
Hello,
when I attempt to use the WebClient example during the connection to the server [line 54] if (client.connect(server, 80)) 
I receive DNS fail (from the debug) yet when I change the  the "server" char pointer to an IP address, connection is made succesfully.

jandrassy, 2020-10-03:
do you have a valid DNS server address?
what is the server name, which fails?

charizord, 2020-10-03:
I have tried multiple servers such as: arduino.cc , google.com , and a random http webpage I found.
I haven't defined a specific DNS server, I presume my home router takes care of that as is with my computer web browser or 
was I supposed to define something specific with the ESP?

This is the debug when I try to connect with  a URL (not ip):
`
Connected to WiFi network.
Starting connection to server...
esp INFO: free linkId is 4
esp INFO: start TCP to arduino.cc:80 on link 4
esp> AT+CIPSTART=4,"TCP","arduino.cc",80 ...sent
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> DNS Fail ...ignored
esp> ERROR ...error
esp ERROR: expected OK got ERROR

disconnecting from server.
`

jandrassy, 2020-10-03:
did you try the WebClient example?

charizord, 2020-10-03:
Like I said, I am using the WebClient example other than these changes : 
-the rx/tx pins of the software serial to pins 2/3
-the arduino serial baud rate to 9600

this is my firmware data:
AT version:1.7.4.0(May 11 2020 19:13:04) 
SDK version:3.0.4(9532ceb)
compile time:May 27 2020 10:12:17
Bin version(Wroom 02):1.7.4 

this is my persistent wifi settings:
Station MAC: 2C:F4:32:XX:XX:XX
hostname ESP-80FD0A
SSID: XXXXXX
BSSID: B0:7F:B9:XX:XX:XX
RSSI:-76 dBm
DHCP: enabled
IP Address: 192.168.XX.XX
gateway IP Address: 192.168.XX.XX
subnet IP mask: 255.255.255.0
**### DNS server: 255.255.255.255** <---- I noticed this is should be 1.1.1.1 in my country, could this be my issue? If yes how do I defined the DNS server in the persistent settings?

jandrassy, 2020-10-03:
in a home local network with router the DNS server is usually the router. but with DHCP the router should send the right DNS server IP.

this is not a problem of this library and I guess even not a problem of the AT firmware.

closed 2020-10-03 by jandrassy