JAndrassy/WiFiEspAT #24: RE: DNS fail

charizord, 2020-10-03:
Hello, truly sorry to be a bother.
I found a solution to my issue with the dns fail, it seems that the "SetupPersistentWiFiConnection" sketch
did not save the DNS addresses received by the router I had to add 

IPAddress DNS = WiFi.dnsServer1();
IPAddress DNS2 = WiFi.dnsServer2();
WiFi.setDNS(DNS,DNS2);

for it to work, I do not know if this is an issue with my ESP8266 model / firmware but I thought it would interest you to know.

Now everything works fine, thank you for the great library and your time.

jandrassy, 2020-10-04:
it is strange that you had to set the DNS IP addresses back. maybe there is a problem in the library.
could you please provide AT commands log from SetupPersistentWiFiConnection and the code above so I can  diagnose the problem?

nelgi, 2020-10-10:
I had exact same issue, I did a re-flash for the ESP-01 (AT 1.7.4) and AT+RESTORE, and now persistent mode works flawless. It might have been sufficient to only do AT+RESTORE.

closed 2020-10-23 by jandrassy