JAndrassy/WiFiEspAT #25: HTTP 400 Bad Request when ussing https

iactiva, 2020-10-08:
Hello,
I'm using this library for posting some data to a url however I'm getting HTTP400 when using ssl port 443 but it is working correctly if using non-secure port 80.
Web server is nginx using Let's Encrypt SSL certificate.
My module is a ESP01 just updated to JB AT 1.7 firmware

I've tried both 
WiFiSSLClient client > client.connect(hostname, 443)
and 
WiFiClient client > client.connectSSL(hostname, 443)

I'm currently posting data to the same server using WiFiNINA library in some other modules w/o issues.
Any suggestion on how to find out a solution?
Thank you!

jandrassy, 2020-10-08:
HTTP 400 is "Bad request". it is a response from the server so connect() was successful. 
400 is usually an error with HTTP protocol, not with transport.

iactiva, 2020-10-08:
Thank you for your reply jandrassy.
I've noticed something weird. Testing your samples with different hosts
WiFiSSLClient works for Hostname api.github.com GET /repos/jandrassy/WiFiEspAT/commits/master/status
However it returns HTTP 400 for Hostname arduino.cc GET /asciilogo.txt (https://www.arduino.cc/asciilogo.txt)

jandrassy, 2020-10-08:
arduino.cc has the updated TLS, which doesn't work with AT firmware only with Jiri Bilek's special firmware.
did you connect to arduino.cc or to www.arduino.cc?

iactiva, 2020-10-08:
I have all modules flashed with with Jiri Bilek's latest firmware v.0.2.3
Connecting to arduino.cc returns HTTP 400 however www.arduino.cc works!

I'm playing now with AT commands and noticed this works properly:

AT+CIPMUX=0
AT+CIPSTART="SSL","arduino.cc",443
AT+CIPSEND=49
GET /asciilogo.txt HTTP/1.1
Host: arduino.cc

jandrassy, 2020-10-09:
show me the sketch which has 400 for arduino.cc

iactiva, 2020-10-09:
Sure. This is the provided WiFiSSLClient.ino sample just replacing hostname and GET path.

#include <WiFiEspAT.h>

const char* server = "arduino.cc";

WiFiSSLClient client;

void setup() {
  Serial.begin(115200);
  while (!Serial);

  Serial1.begin(115200);
  WiFi.init(Serial1);

  if (WiFi.status() == WL_NO_MODULE) {
    Serial.println();
    Serial.println("Communication with WiFi module failed!");
    // don't continue
    while (true);
  }

  // waiting for connection to Wifi network set with the SetupWiFiConnection sketch
  Serial.println("Waiting for connection to WiFi");
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.print('.');
  }
  Serial.println();
  Serial.println("Connected to WiFi network.");

  Serial.println("Starting connection to server...");
  if (client.connect(server, 443)) { // port 443 is the default https port
    Serial.println("connected to server");

    client.println("GET /asciilogo.txt HTTP/1.1");
    client.print("Host: ");
    client.println(server);
    client.println("User-Agent: Arduino");
    client.println("Connection: close");
    client.println();
    client.flush();
  }
}

void loop() {

  // if there are incoming bytes available
  // from the server, read them and print them
  while (client.available()) {
    char c = client.read();
    Serial.write(c);
  }

  // if the server's disconnected, stop the client
  if (!client.connected()) {
    Serial.println();
    Serial.println("disconnecting from server.");
    client.stop();

    // do nothing forevermore
    while (true);
  }
}

Here is the output HTTP 400:
Waiting for connection to WiFi
.
Connected to WiFi network.
Starting connection to server...
connected to server
<html>
<head><title>400 Bad Request</title></head>
<body bgcolor="white">
<center><h1>400 Bad Request</h1></center>
<hr><center>nginx</center>
</body>
</html>

disconnecting from server.

jandrassy, 2020-10-11:
could you turn on logging?
https://github.com/jandrassy/WiFiEspAT#logging

iactiva, 2020-11-09:
Hi @jandrassy,
Sorry for the big delay in answering. It took me some time to stock some new ESP-01 modules. There seems this issue was something telated to those old modules as these new ones seem to be working as expected... so I'm closing this issue.
Many thanks for your support.

closed 2020-11-09 by iactiva