JAndrassy/WiFiEspAT #27: Your Serial.println commands not work!

makarna35, 2020-10-10:
Hi;

Your Serial.println commands in your examples can't print any data. I double check maybe your library have more BUGS
I use Arduino IDE version: 1.8.13

jandrassy, 2020-10-11:
turn on debug prints
https://github.com/jandrassy/WiFiEspAT#logging

makarna35, 2020-10-11:
Print available networks example work. I can see my SSID (wpa2) in list.
I'm set log level to LOG_LEVEL_DEBUG

output is:
`esp INFO: soft reset
esp> AT+RST ...sent
esp> AT+RST ...ignored
esp> OK ...ignored
esp INFO: soft reset
esp> AT+RST ...sent
esp> AT+RST ...ignored
esp> OK ...ignored
esp> bB׆⸮Sbj⸮⸮ȤRN⸮ȤRN⸮H⸮⸮NO<⸮⸮ ...ignored
esp> ⸮⸮⸮⸮⸮.Db&t⸮5⸮֖⸮C⸮V⸮P⸮⸮ ...ignored
esp> ready ...matched
esp> ATE0 ...sent
esp> ATE0 ...ignored
esp> OK ...matched
esp> AT+CIPMUX=1 ...sent
esp> OK ...matched
esp> AT+CIPRECVMODE=1 ...sent
esp> OK ...matched
esp> AT+CWMODE? ...sent
esp> +CWMODE:1 ...matched
esp> OK ...matched
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:5 ...matched
esp> OK ...matched
Waiting for connection to WiFi
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:5 ...matched
esp> OK ...matched
.esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:5 ...matched
esp> OK ...matched
`
Still can't connect to SSID

jandrassy, 2020-10-11:
what sketch do you run?

makarna35, 2020-10-11:
I use modified WiFi Web Server file by me:

`/*
  WiFi Web Server

 A simple web server that shows the value of the analog input pins.

 created 13 July 2010
 by dlf (Metodo2 srl)
 modified 31 May 2012
 by Tom Igoe
 modified in Jul 2019 for WiFiEspAT library
 by Juraj Andrassy https://github.com/jandrassy
 */

#include <WiFiEspAT.h>

#include "arduino_secrets.h"

///////please enter your sensitive data in the Secret tab/arduino_secrets.h
const char ssid[] = SECRET_SSID;    // your network SSID (name)
const char pass[] = SECRET_PASS;    // your network password (use for WPA, or use as key for WEP)

// Emulate Serial1 on pins 6/7 if not present
#if defined(ARDUINO_ARCH_AVR) && !defined(HAVE_HWSERIAL1)
#include <SoftwareSerial.h>
SoftwareSerial Serial1(6, 7); // RX, TX
#define AT_BAUD_RATE 9600
#else
#define AT_BAUD_RATE 115200
#endif

WiFiServer server(80);

void setup() {

  Serial.begin(115200);
  while (!Serial);

  setEspBaudRate(AT_BAUD_RATE);
  
  WiFi.init(Serial1);

  if (WiFi.status() == WL_NO_MODULE) {
    Serial.println("Communication with WiFi module failed!");
    // don't continue
    while (true);
  }

  // waiting for connection to Wifi network set with the SetupWiFiConnection sketch
  Serial.println("Waiting for connection to WiFi");
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.print('.');
  }
  Serial.println();

  server.begin();

  IPAddress ip = WiFi.localIP();
  Serial.println();
  Serial.println("Connected to WiFi network.");
  Serial.print("To access the server, enter \"http://");
  Serial.print(ip);
  Serial.println("/\" in web browser.");
}

void loop() {

  WiFiClient client = server.available();
  if (client) {
    IPAddress ip = client.remoteIP();
    Serial.print("new client ");
    Serial.println(ip);

    while (client.connected()) {
      if (client.available()) {
        String line = client.readStringUntil('\n');
        line.trim();
        Serial.println(line);

        // if you've gotten to the end of the HTTP header (the line is blank),
        // the http request has ended, so you can send a reply
        if (line.length() == 0) {
          // send a standard http response header
          client.println("HTTP/1.1 200 OK");
          client.println("Content-Type: text/html");
          client.println("Connection: close");  // the connection will be closed after completion of the response
          client.println("Refresh: 5");  // refresh the page automatically every 5 sec
          client.println();
          client.println("<!DOCTYPE HTML>");
          client.println("<html>");
          // output the value of analog input pins
          for (int analogChannel = 0; analogChannel < 4; analogChannel++) {
            int sensorReading = analogRead(analogChannel);
            client.print("analog input ");
            client.print(analogChannel);
            client.print(" is ");
            client.print(sensorReading);
            client.println("<br />");
          }
          client.println("</html>");
          client.flush();
          break;
        }
      }
    }

    // close the connection:
    client.stop();
    Serial.println("client disconnected");
  }
}

void setEspBaudRate(unsigned long baudrate){
  long rates[6] = {115200,74880,57600,38400,19200,9600};

  Serial.print(F("ESP8266 baudrate "));
  Serial.print(baudrate);
  Serial.print(F("'e ayarlanıyor"));
  Serial.println(F("..."));

  for(int i = 0; i < 6; i++){
    Serial1.begin(rates[i]);
    delay(100);
    Serial1.print(F("AT+UART_DEF="));
    Serial1.print(baudrate);
    Serial1.print(F(",8,1,0,0\r\n"));
    delay(100);  
  }
    
  Serial1.begin(baudrate);
}`

my arduino_secrets.h file have correct WiFi ssID and SSID password.

makarna35, 2020-10-11:
OK.

I'm add: `int status = WiFi.begin(ssid, pass);`

Now can connect to SSID.
Thanks.

closed 2020-10-11 by jandrassy