JAndrassy/WiFiEspAT #30: MQTT

abhay9sap, 2020-10-28:
Hi,
Your post in Arduino Forum helped me flash firmware of ESP8266 on my ATMega Wifi with ESP8266.

I tried using WiFiEsp.h but I am facing issues with new firmware and your library helps solve that.

Can you help me with Example file for Mqtt connection, I tried with PubSubClient but I get PubSubClient error rc=-1, connection works with WiFiEsp.h.

Thank you,

Regards,
Abhay

My Code is 
```
#include <WiFiEspAT.h>
#include <PubSubClient.h> 
IPAddress server(192, 168, 10, 3); 
WiFiClient espClient;
PubSubClient client(espClient); 
int status = WL_IDLE_STATUS;

#include "arduino_secrets.h"
///////please enter your sensitive data in the Secret tab/arduino_secrets.h
const char ssid[] = SECRET_SSID;    // your network SSID (name)
const char pass[] = SECRET_PASS;    // your network password (use for WPA, or use as key for WEP)

// Emulate Serial1 on pins 6/7 if not present
//#if defined(ARDUINO_ARCH_AVR) && !defined(HAVE_HWSERIAL1)
//#include "SoftwareSerial.h"
//SoftwareSerial Serial1(6, 7); // RX, TX
//#define AT_BAUD_RATE 9600
//#else
//#define AT_BAUD_RATE 115200
//#endif

void setup() {
Serial.begin(9600);
Serial3.begin(115200);
  //while (!Serial);

  //Serial3.begin(AT_BAUD_RATE);
WiFi.init(Serial3);

 if (WiFi.status() == WL_NO_MODULE) {
    Serial.println();
    Serial.println("Communication with WiFi module failed!");
    // don't continue
    while (true);
  }

  WiFi.disconnect(); // to clear the way. not persistent

  WiFi.setPersistent(); // set the following WiFi connection as persistent

  WiFi.endAP(); // to disable default automatic start of persistent AP at startup

//  uncomment this lines for persistent static IP. set addresses valid for your network
//  IPAddress ip(192, 168, 1, 9);
//  IPAddress gw(192, 168, 1, 1);
//  IPAddress nm(255, 255, 255, 0);
//  WiFi.config(ip, gw, gw, nm);

  Serial.println();
  Serial.print("Attempting to connect to SSID: ");
  Serial.println(ssid);

//  use following lines if you want to connect with bssid
//  const byte bssid[] = {0x8A, 0x2F, 0xC3, 0xE9, 0x25, 0xC0};
//  int status = WiFi.begin(ssid, pass, bssid);

  int status = WiFi.begin(ssid, pass);

  if (status == WL_CONNECTED) {
    Serial.println();
    Serial.println("Connected to WiFi network.");
    printWifiStatus();
  } else {
    WiFi.disconnect(); // remove the WiFi connection
    Serial.println();
    Serial.println("Connection to WiFi network failed.");
  }
  //connect to MQTT server 
client.setServer(server, 1883); 
client.setCallback(callback); 
client.subscribe("home/led1");
}

//print any message received for subscribed topic
void callback(char* topic, byte* payload, unsigned int length) { 
Serial.print("Message arrived ["); 
Serial.print(topic); 
Serial.print("] ");
 for (int i=0;i<length;i++) { 
Serial.print((char)payload[i]); 
} 
Serial.println(); 
} 

void loop() {
   // put your main code here, to run repeatedly:
 if (!client.connected()) { 
reconnect();
 }
 client.loop(); 
}

void printWifiStatus() {

  // print the SSID of the network you're attached to:
  char ssid[33];
  WiFi.SSID(ssid);
  Serial.print("SSID: ");
  Serial.println(ssid);

  // print the BSSID of the network you're attached to:
  uint8_t bssid[6];
  WiFi.BSSID(bssid);
  Serial.print("BSSID: ");
  printMacAddress(bssid);

  uint8_t mac[6];
  WiFi.macAddress(mac);
  Serial.print("MAC: ");
  printMacAddress(mac);

  // print your board's IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
}

void printMacAddress(byte mac[]) {
  for (int i = 5; i >= 0; i--) {
    if (mac[i] < 16) {
      Serial.print("0");
    }
    Serial.print(mac[i], HEX);
    if (i > 0) {
      Serial.print(":");
    }
  }
  Serial.println();
}

void reconnect() { 
// Loop until we're reconnected
 while (!client.connected()) {
 Serial.print("Attempting MQTT connection..."); 
 Serial.print(server);
// Attempt to connect, just a name to identify the client
 if (client.connect("arduinoClient")) {
 Serial.println("connected"); // Once connected, publish an announcement... 
client.publish("home/led1","1"); // ... and resubscribe 
client.subscribe("home");
client.subscribe("home/led1");
 } else {
 Serial.print("failed, rc=");
 Serial.print(client.state());
 Serial.println(" try again in 5 seconds");
 // Wait 5 seconds before retrying
 delay(5000);
 }
 }
 }
```

jandrassy, 2020-10-28:
you could turn on debug log.
https://github.com/jandrassy/WiFiEspAT#logging

where do you get rc -1?

abhay9sap, 2020-10-30:
Hi Jandrassy,

sorry for late reply, I got the rc=-1 on Serial Monitor from reconnect() part of code client.state().

I check debug log and report back.

jandrassy, 2020-10-30:
replace
#define WIFIESPAT_LOG_LEVEL LOG_LEVEL_SILENT
with
#define WIFIESPAT_LOG_LEVEL LOG_LEVEL_DEBUG

abhay9sap, 2020-10-31:
Hi Jandrassy,
[Logfile.txt](https://github.com/jandrassy/WiFiEspAT/files/5469160/Logfile.txt)

I have attached log file

Thank you :)

jandrassy, 2020-10-31:
you use some older version of the library?

abhay9sap, 2020-11-02:
Hi Jandrassy,

I think both are latest version,

Following are the library version
PubSubClient 2.8.0
WifiEspAT 1.3.1

Thank you :)

jandrassy, 2020-11-03:
```
esp> AT+SYSSTORE=0 ...sent
esp> ERROR ...error
esp ERROR: expected OK got ERROR
esp WARN: Error setting store mode. Is the firmware AT2?
```

abhay9sap, 2020-11-04:
Hi Jandrassy,

I thought as my AT Firmware was 3.0.4 I had to follow this Instruction;

> For AT firmware version 2.1.0 or higher open in the folder of the library the file src/utility/EspAtDrvTypes.h in a text editor and comment out the line #define WIFIESPAT1 like this //#define WIFIESPAT1

I now removed the comment out and its working fine, Thank you so much for helping me out on this.

Regard
Abhay 👍

closed 2020-11-04 by abhay9sap

jandrassy, 2020-11-04:
AT firmware version is different from SDK version. 
current non-os SDK 3+ AT versions are 1.7+. 
rtos SDK AT fw is version 2+

abhay9sap, 2020-11-06:
Hi Jandrassy,

Thank you for that information, I will look out for AT version on my future project.

But your Library is just awesome. Testing it for last 2 days, its rock solid and functions flawlessly.

Thank you again for helping the community and wish you Good Luck.

Regards,
Abhay

leejames92, 2021-08-14:
Hi, might be a newbie question to ask but how do I enable logging specifically? Because I'm having MQTT disconnect after some period of time and it's driving me crazy as I do not know why :(

leejames92, 2021-08-14:
Oh I got it to work and show debug in my serial monitor. Is it able to automatic export into a .txt file?

Thank you.