JAndrassy/WiFiEspAT #31: disable some functions to save flash space

atesin, 2020-10-29:
hi... i am a fan of your plugin, i use it daily xD

is there any way to disable some unused functions to save space to other stuff in sketch?...

for example, i never use ssl or udp and my sketch sizes about 95% of flash, i would like that free space to put a couple more commands, manual or web pages (telnet or http server)...  currently i feel somewhat forced to get and put some sdcard module and library that will complicate and grow my config and sketch anyway

jandrassy, 2020-10-29:
95% is OK it doesn't cause problems, but SD library will not fit into that 5% remaining.
the compiler optimizes unused code away. so udp and ssl doesn't take any place in your sketch if you don't use them.
I assume logging is off.

atesin, 2020-10-29:
i currently have an irrigation system configurable by telnet commands (with some man pages)...  i would like to turn into an http server for easy usage, but http uses more text :(

anyway thanks for your time, response and effort

(still the best esp8266 library!)

jandrassy, 2020-10-29:
ATmega328p is small for IoT

closed 2020-10-29 by jandrassy