JAndrassy/WiFiEspAT #33: MQTT connection...failed, rc=-2 try again in 5 seconds

Muplex, 2020-12-05:
Hello,
I hope you can help me with these error.
All seems to run smooth for a random ammount of time, then MQTT loose connectivity and did not connect anymore.

I set debug level 4 in  src/utils/EspAtDrvLogging.h having these message:

`Connecting to MY_SEED
Attempting to connect to WPA SSID: MY_SEED
esp INFO: join AP MY_SEED current

WiFi connected
IP address: 
esp INFO: STA IP query
192.168.178.114
Attempting MQTT connection...esp INFO: free linkId is 4
esp INFO: start TCP to 192.168.178.175:1883 on link 4
esp INFO: BuffManager new buff.stream at 0 for linkId 4 rx 64 tx 64
esp INFO: sync
esp INFO: send data on link 4
esp INFO: 	sent 19 bytes on link 4
esp INFO: get data on link 4
esp INFO: 	got 4 bytes on link 4
MQTT connected
esp INFO: send data on link 4
esp INFO: 	sent 64 bytes on link 4
esp INFO: send data on link 4
esp INFO: closed linkId 4
esp ERROR: failed to send data
Attempting MQTT connection...esp INFO: free linkId is 3
esp INFO: start TCP to 192.168.178.175:1883 on link 3
esp ERROR: expected OK got ERROR
failed, rc=-2 try again in 5 seconds
Attempting MQTT connection...esp INFO: closed linkId 3
esp INFO: free linkId is 3
esp INFO: start TCP to 192.168.178.175:1883 on link 3
esp ERROR: expected OK got ERROR
failed, rc=-2 try again in 5 seconds
......
`

jandrassy, 2020-12-06:
the "failed" message is from some MQTT library.
this looks like INFO level logging. set DEBUG
`#WIFIESPAT_LOG_LEVEL LOG_LEVEL_DEBUG`

Muplex, 2020-12-06:
oh sorry, previously made wrong copy-paste

this is 

`

Connecting to MY_SSID
Attempting to connect to WPA SSID: MY_SSID
esp INFO: join AP ZEDGEAR50 current
esp> AT+CWJAP_CUR="MY_SSID","MY PASS" ...sent
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> WIFI CONNECTED ...ignored
esp> WIFI GOT IP ...ignored
esp> OK ...matched

WiFi connected
IP address: 
esp INFO: STA IP query
esp> AT+CIPSTA? ...sent
esp> +CIPSTA:ip:"192.168.178.114" ...matched
esp> +CIPSTA:gateway:"192.168.178.1" ...matched
esp> +CIPSTA:netmask:"255.255.255.0" ...matched
esp> OK ...matched
192.168.178.114
Attempting MQTT connection...esp INFO: free linkId is 4
esp INFO: start TCP to 192.168.178.175:1883 on link 4
esp> AT+CIPSTART=4,"TCP","192.168.178.175",1883 ...sent
esp> 4,CONNECT ...ignored
esp> OK ...matched
esp INFO: BuffManager new buff.stream at 0 for linkId 4 rx 64 tx 64
esp INFO: sync
esp> AT+CIPSTATUS ...sent
esp> STATUS:3 ...matched
esp> +CIPSTATUS:4,"TCP","192.168.178.175",1883,13486,0 ...matched
esp> OK ...end of list
esp> AT+CIPRECVLEN? ...sent
esp> +CIPRECVLEN:0,0,0,0,0 ...matched
esp> OK ...matched
esp INFO: send data on link 4
esp> AT+CIPSEND=4,19 ...sent

UNTILL NOW IS WORKING WELL...
THEN START THE PROBLEM 

esp> AT+CIPSTATUS ...sent
esp> STATUS:3 ...matched
esp> +CIPSTATUS:4,"TCP","192.168.178.175",1883,13486,0 ...matched
esp> OK ...end of list
esp> AT+CIPRECVLEN? ...sent
esp> +CIPRECVLEN:0,0,0,0,0 ...matched
esp> OK ...matched
esp INFO: send data on link 4
esp> AT+CIPSEND=4,2 ...sent
esp> OK ...ignored
esp> > ...matched
esp> Recv 2 bytes ...matched
esp> ?
esp> busy s... ...ignored
esp> ?
esp> busy s... ...ignored
esp> ?
esp> busy s... ...ignored
esp> ?
esp> busy s... ...ignored
esp> ?
esp> busy s... ...ignored
esp> 4,CLOSED ...processed
esp INFO: closed linkId 4
esp> SEND FAIL ...matched
esp ERROR: failed to send data
Attempting MQTT connection...esp INFO: free linkId is 4
esp INFO: start TCP to 192.168.178.175:1883 on link 4
esp> AT+CIPSTART=4,"TCP","192.168.178.175",1883 ...sent
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ERROR ...error
esp ERROR: expected OK got ERROR
failed, rc=-2 try again in 5 seconds
Attempting MQTT connection...esp> 4,CLOSED ...processed
esp INFO: closed linkId 4
esp INFO: free linkId is 4
esp INFO: start TCP to 192.168.178.175:1883 on link 4
esp> AT+CIPSTART=4,"TCP","192.168.178.175",1883 ...sent
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ERROR ...error
esp ERROR: expected OK got ERROR
failed, rc=-2 try again in 5 seconds
`

Muplex, 2020-12-06:
AT+CIPSTART=4

i notice that sometime this number change. Is it the router radio channel ?
A sort of loose of channel could be linked to the problem?

jandrassy, 2020-12-06:
the AT firmware is first unable to sent the data (SEND FAIL) and in the next attempt it is unable to connect.
4 is the linkid of 5 links of AT firmware. this is an AT firmware specific thing. WiFiEspAT uses linkid starting at 4 and counting down (to leave linkids starting at 0 free for incoming connections). so you sometimes have link 3 if 4 is still busy (stuck on sending for example)

it could be a problem of AT firmware, but the 1.7.4 is pretty good and stable so I would blame the server. Do you have some server logs.

Muplex, 2020-12-06:
> it could be a problem of AT firmware, but the 1.7.4 is pretty good and stable so I would blame the server. Do you have some server logs.

I have the 1.7.0 flashed on the ESP8266 of the WiFi mega board.

I try to update it right now and in the meanwhile i will provide you Mosquitto logs.

Thank you for your help

Muplex, 2020-12-06:
This what I found in the mosquitto logs 

`1607246462: New connection from 192.168.178.114 on port 1883.
1607246462: New client connected from 192.168.178.114 as Alarm (p2, c1, k15).
1607246492: New connection from 192.168.178.114 on port 1883.
1607246492: Client Alarm already connected, closing old connection.
1607246492: New client connected from 192.168.178.114 as Alarm (p2, c1, k15).
1607246515: Client Alarm has exceeded timeout, disconnecting.`

this message repeats over and over

is enought or I have to increase debugging level of mosquitto ?

Muplex, 2020-12-06:
I replace another arduino copiing the same schetch...but obviously they are not connected toghether
could be the client Name linked to the problem??

Muplex, 2020-12-06:
> I have the 1.7.0 flashed on the ESP8266 of the WiFi mega board.
> 
> I try to update it right now and in the meanwhile i will provide you Mosquitto logs.

AT version 1.7.4.0 do not solve. same problem and messages on serial

jandrassy, 2020-12-06:
so on AT firmware side there is SEND FAILED and on server there is timeout waiting for data. but then the AT firmware can't connect. some network problem?

Muplex, 2020-12-06:
> so on AT firmware side there is SEND FAILED and on server there is timeout waiting for data. but then the AT firmware can't connect. some network problem?

mmmh, no I think. As told I have other NODE mcu based on ESP on the net, working the same way and other TASMOTA sonoff devices that gives no problems.
Could you recommend me some tests to try to understand the root cause of the problem???
unfortunately I am a hobbyist and many background knowledge is missing to me.

jandrassy, 2020-12-06:
I would use Wireshark on server to see the network traffic. but that is very advanced.

Muplex, 2020-12-07:
Unfortunatelly I didn't found a solution.
I dismount the Mega+wifi and I opeted for a LolinNodeMCU 1.0 + 1 x multiplexer.
I choose that solution because I have several lolin reading reading signal from serveral sensor each that worked for years without any problem.
I tried another board more "switable" for the project but something is missing.

Thank you for your help anycase Mr jandrassy. 
God Bless you

closed 2021-01-06 by jandrassy