JAndrassy/WiFiEspAT #35: timeout doesn't work for WiFiServer.begin(links, timeout)

atesin, 2020-12-13:
default timeout for WiFiServer(linksCount) is 30 seconds

when i create a server with WiFiServer(linksCount, 120) or whatever timeout, it still fixes at 30 seconds

i had to directly send espSerial.println("AT+CIPSTO=120"); and busy-wait for response just after server creation

- arduino uno r3
- esp-01s (esp8266 1mb flash)
- latest lobo firmware, with AT v1.7
- arduino ide 1.8.13
- wifiespat 1.3.1

jandrassy, 2020-12-13:
`server.begin` has connection count and timeout.
with debug prints enabled, is AT+CIPSTO logged?

atesin, 2020-12-14:
not loged.... just observation
i am making  a programable irrigation system with relays, pumps and valves, configurable throught  a telnet server with commands

i noticed it disconnects me in 30 seconds regadless the begin() timeout value, so i tried directly with serial->at+cipsto and works as expected

i will try enabling debug and report again ..  never tried before

jandrassy, 2020-12-14:
could you please turn on debug logging and see what is happening on AT commands level?
https://github.com/jandrassy/WiFiEspAT#logging

atesin, 2020-12-15:
ok i have it, some lines were stripped for sake of clarity,
i define all variable in preproc constants, so is easy to disable entire blocks of code if needed
```
#define LISTEN_TCP_PORT 1234
#define ESP_LINK_TIMEOUT 600
int listenPort = LISTEN_TCP_PORT;
int linkTimeout = ESP_LINK_TIMEOUT;
WiFiServer server(listenPort);
server.begin(WIFIESPAT_LINKS_COUNT, linkTimeout); // max_links, client_timeout
```
... meanwhile in serial console...
```
esp INFO: begin server at port 1234
esp> AT+CIPSERVERMAXCONN=5 ...sent
esp> OK ...matched
esp> AT+CIPSERVER=1,1234 ...sent
esp> OK ...matched
esp> AT+CIPSTO=88 ...sent
esp> OK ...matched
```
AT+CIPSTO=88 ???  o_Ô

jandrassy, 2020-12-15:
but this was fixed in 1.3.1

atesin, 2020-12-16:
then sorry but i don't know why is working this way..... i have 1.3.1
in what commit was fixed?, what files/lines? ...

jandrassy, 2020-12-16:
> then sorry but i don't know why is working this way..... i have 1.3.1
> in what commit was fixed?, what files/lines? ...

https://github.com/jandrassy/WiFiEspAT/commit/15f853fb5c32e90105d5fc02eab99f91d22b5f6b#diff-0db127cd7416d8da26968bb023a0359e5890a25728d2dc6a0f140e3bcf17169dR28

atesin, 2020-12-17:
i see...

but why i have these lines if i supposedly have v1.3.1? , what could had happen?

my src/WiFiServer.cpp file line 28
```
void WiFiServer::begin(uint8_t maxConnCount, uint8_t serverTimeout) {
  state = EspAtDrv.serverBegin(port, maxConnCount, serverTimeout) ? LISTEN : CLOSED;
}
```

jandrassy, 2020-12-18:
I just downloaded the Library Manager version 1.3.1 and it has `uint16_t serverTimeout`

closed 2020-12-18 by jandrassy

atesin, 2020-12-19:
strange  ... maybe because i had patched files the library manager got confused with hashes/versions/updates :/

i will uninstall, reinstall and repatch  (another reason add getLinkId() in official reeases ??  xD )

thanks for your time and effort, your library is fantastic!

atesin, 2020-12-23:
hi i know this issue is closed, but i had difficults with library versions and reinstalliation,, and want to share what i had found

as arduino documentantion says i went to <myDocs>/arduino/libraries/ and deleted the whole WiFiEspAt folder, however when i restart the ide the library was still in library manager, even the examples were present and even my sketches compiled fine...  so i as the lib were still present wasn't able to remove and install the lib again from the scratch

so i found there was a COPY of the library folder named like "arduino_4234234", so i deleted also and when restarted the ide the library gone and could reinstall (and patch) it again... maybe that is why the version mismatch also

i wish to share this issue if somebody else have a similar problem

jandrassy, 2020-12-24:
this happens if the IDE can't delete the folder of the library while installing a new version of the library