JAndrassy/WiFiEspAT #36: Question about extending the command set

JiriBilek, 2020-12-30:
Hi Juraj,
I started to use your library and I like it, it's quite stable. I run it for couple of weeks in 24/7 mode.
But I use my AT firmware for ESP with extended command set (https://github.com/JiriBilek/ESP_ATMod).
And the question is: what's the best way to extend the command set of your library? My first choice was the `EspAtDrvClass` object but several important functions are private. When functions as `sendData `are public, why `sendCommand `and `simpleCommand `are kept private?

jandrassy, 2020-12-30:
odpisem slovensky, lebo aj tak to adresovane len tebe Jiri. nemam problem pridat nestandardne AT prikazy pre tvoj fw. ono je to divne, ze tam je ta AT vrstva, ale inak je to celkom sikovna kombinacia kniznice a firmwaru, ktory je Arduino sketch.
takze kludne dopln do EspAtDrv alebo este lepsie mozno, ze ja na slepo doplnim co dohodneme a ty to potom odladis s fw.

sendCommand a simpleCommand su privatne lebo sa pouzivaju len v ramci triedy. to len pre poriadok

JiriBilek, 2020-12-30:
Ahoj,
mě přišlo, nejrozumnější přidat co nejuniverzálnější funkce. Nemá cenu doplnit funkce vysloveně šité na můj fw, když to nebude úplně nutné.
Část bych určitě zvládl pomocí zveřejnění `sendCommand`. Teď v tento okamžik mě pálí nejvíc příkaz SYSTIME, protože potřebuju na klientovi vědět čas:
```
AT+SYSTIME?
+SYSTIME:1607438042
OK
```
Uděláme to tak, že si upravím tvoji knihovnu a pošlu to jako PR nebo Issue. Nechce se mi forknout tvoji knihovnu, protože se nechci starat o update (už takhle nevím, kde mi hlava stojí).

Díky moc a měj se fajn.
J.

EN: Sorry guys, this theme is about specific changes to the library and we took advantage we can understand each other language :)

jandrassy, 2020-12-30:
je tam sntpTime() pre WiFi.getTime() co som robil pre AT Lobo. AT+SNTPTIME?

JiriBilek, 2020-12-30:
To jsem si nevšiml, že AT Lobo to má takhle. Tak to u sebe upravím, ať máme stejný příkaz.
Dík.

closed 2020-12-31 by jandrassy

jandrassy, 2021-01-03:
oficialny AT 1.7 ma AT+CIPSNTPTIME? ten ale vrati cas ako formatovany string `Thu	Aug	04	14:48:05	2016`, takze som to nepouzil.
AT 2.1 ma AT+SYSTIMESTAMP? , ktory vrati epoch