JAndrassy/WiFiEspAT #39: Update  remote database Xamp server

sgamoct, 2021-02-21:
Hi, I am trying to insert data on a remote database which is built on xamp.
unfortunately I can't write these two data. I'm going crazy I don't understand where the problem is. 
the wifi is connected and from the pc or mobile phone connected to the same network I can write and read the database.
this is my code:

`  void loop() {
  WiFiClient  client;
  strURL ="GET 192.168.0.131/sudatabase/admin/system/functions.php?Red=xyz&H=69 HTTP/1.1";
  client.println (strURL);
  client.println("Connection: close");
  client.println();
  client.stop();
  delay (5000);
  Serial.println("inviato");
  }`

the server has a local address of 192.168.0.131 and is installed on xampp and can be reached by other devices.
can you help me to solve this problem?

jandrassy, 2021-02-21:
see the WiFiClient example

sgamoct, 2021-02-21:
> guarda l'esempio WiFiClient

WebClient?

jandrassy, 2021-02-21:
> > guarda l'esempio WiFiClient
> 
> WebClient?

sorry, yes, WebClient

sgamoct, 2021-02-21:
Ok i try it.
thankyou

closed 2021-02-21 by jandrassy

sgamoct, 2021-02-21:
Happy, work fine!
thankyou