JAndrassy/WiFiEspAT #41: undefined reference to `typeinfo for arduino::Print'

Dracanrage, 2021-03-17:
When I try to define variable of class WiFiClient, it shows undefined reference to `typeinfo for arduino::Print'.
[image]

jandrassy, 2021-03-17:
it is an Arduino library

closed 2021-03-17 by jandrassy

Dracanrage, 2021-03-17:
> it is an Arduino library

Yes, I used this library, it seems works well without use WiFiEspAT Lib, but I don't know why it turns wrong when I try to compile it through CMake (I used VSCode).

jandrassy, 2021-03-17:
WiFiEspAT is an Arduino library. it requires an Arduino core for the selected MCU architecture.  it is intended to be build by Arduino builder