JAndrassy/WiFiEspAT #45: Time_Zone is not working function WiFi.sntp(TIME_ZONE, "us.pool.ntp.org");

tonbor, 2021-05-01:
Changed nothing to SNTPtime.ino 
Get Always the UTC time, no matter what number I put in  const int8_t TIME_ZONE = 2; // UTC + 2
-14 or 13 always UTC time

Waiting for SNTP
.
2021-05-01 09:15:57
2021-05-01 09:15:58
2021-05-01 09:15:59
2021-05-01 09:16:00
2021-05-01 09:16:01
2021-05-01 09:16:02
2021-05-01 09:16:03
2021-05-01 09:16:04
2021-05-01 09:16:05
2021-05-01 09:16:06
2021-05-01 09:16:07
2021-05-01 09:16:08
ESP 01 8bit
OK
AT+GMR
AT version:2.2.0.0-dev(ecf20b7 - Jul 24 2020 08:05:55)
SDK version:v3.3-2-gb6e861a
compile time(7650c83):Aug  1 2020 20:15:06
Bin version:2.0.0(Unknown)

OK

jandrassy, 2021-05-01:
yes, sorry. the timezone parameter is not used. I will remove it.  the AT command AT+CIPSNTPCFG has the parameter, but the AT firmware uses it only for formatted timestamp. 
getTime() returns epoch seconds which must be UTC

closed 2021-05-01 by jandrassy

tonbor, 2021-05-01:
Thanks for the Quick fix.