JAndrassy/WiFiEspAT #46: SSL connection question

NorbertSandor, 2021-05-05:
I tried the WiFiSSLClient example from Arduino (almost unchanged, only changed Serial1 to Serial3).
My problem is that the HTTP response is not received:

```
Waiting for connection to WiFi
......
Connected to WiFi network.
Starting connection to server...
connected to server

disconnecting from server.
```

The firmware seems to be OK based on the CheckFirmware example:

```
AT firmware version 1.7.4.0
AT firmware is OK for the WiFiEspAT library.
```

Is there any additional configuration I'm missing? Thanks.

NorbertSandor, 2021-05-05:
Normal ```http``` protocol seems to work, the (slightly modified*) WebClient example prints the expected HTTP response to Serial.

* the ```WiFi.begin(SECRET_SSID, SECRET_PASS);``` part is missing from the WebClient example

jandrassy, 2021-05-06:
with 1.7.4 SSL is not supported. sorry I forgot to write that in the example.
https://github.com/jandrassy/WiFiEspAT#limitations

NorbertSandor, 2021-05-06:
Thanks for the info.
So I have to upgrade to AT 2.1 to use SSL.
I have already tried to upgrade several times but always failed (the ESP8266 became unresponsive). Maybe do you know some source/article/documentation about the AT 2.1 upgrade process?

jandrassy, 2021-05-06:
https://github.com/jandrassy/WiFiEspAT#at-21-1

NorbertSandor, 2021-05-06:
Unfortunatelly, it did not work for me :(
I have a somewhat special board, maybe it's the reason: https://www.gabrielcsapo.com/arduino-web-server-mega-2560-r3-built-in-esp8266/
Thanks for your help.

jandrassy, 2021-05-06:
what version of AT 2.1 did you upload and what are the parameters of flash memory on your esp8266.
at default the AT 2.1 communicates on pins 15 as TX and 13 as RX.

NorbertSandor, 2021-05-06:
> what version of AT 2.1 did you upload and what are the parameters of flash memory on your esp8266.

V2.1.0.0
32Mb

>  at default the AT 2.1 communicates on pins 15 as TX and 13 as RX

I don't know if this is different from the pin configuration of the AT 1.7 firmware. If yes then this may be the cause.

Besides, I could make secure connections to work using the _JB AT 1.7_ firmware.
Unfortunatelly it has limited functionality so it is not ideal :\

jandrassy, 2021-05-07:
read this again. slow
https://github.com/jandrassy/WiFiEspAT#at-21-1

NorbertSandor, 2021-05-07:
"slow" - :D
Thanks for your patience ;)

With your instructions I have uploaded the AT 2.1 firmware successfully.

Using direct serial connection to the ESP8266, I was able to connect to my network:
```
...
AT+CWJAP="AP_NAME","***"
WIFI CONNECTED
```

But I'm unable to connect using WiFiEspAT :(
I used the modification of your description:
```
#ifndef WIFIESPAT2 // for -D
// #define WIFIESPAT1
#endif
```
(Is this correct?)

Sketch:
```
#include <WiFiEspAT.h>

const char ssid[] = "AP_NAME";
const char pass[] = "***";
#define AT_BAUD_RATE 115200

void printMacAddress(byte mac[])
{
  for (int i = 5; i >= 0; i--)
  {
    if (mac[i] < 16)
    {
      Serial.print("0");
    }
    Serial.print(mac[i], HEX);
    if (i > 0)
    {
      Serial.print(":");
    }
  }
  Serial.println();
}

void printWifiStatus()
{
  char ssid[33];
  WiFi.SSID(ssid);
  Serial.print("SSID: ");
  Serial.println(ssid);

  uint8_t bssid[6];
  WiFi.BSSID(bssid);
  Serial.print("BSSID: ");
  printMacAddress(bssid);

  uint8_t mac[6];
  WiFi.macAddress(mac);
  Serial.print("MAC: ");
  printMacAddress(mac);

  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
}

void setup()
{
  Serial.begin(115200);
  while (!Serial);

  Serial3.begin(AT_BAUD_RATE);
  WiFi.init(Serial3);

  int status = WiFi.begin(ssid, pass);

  if (status == WL_CONNECTED)
  {
    Serial.println();
    Serial.println("Connected to WiFi network.");
    printWifiStatus();
  }
  else
  {
    WiFi.disconnect(); // remove the WiFi connection
    Serial.println();
    Serial.println("Connection to WiFi network failed.");
  }
}

void loop()
{
}
```

Serial output with logging enabled:
```
esp INFO: soft reset  
esp> AT+RST ...sent   
esp> AT+RST ...ignored
esp> OK ...ignored    
esp> rl ...ignored
esp> ␌␌�␌l␌␌␌␌␌␌c␌o�|␃�␌␌␌c��on� ...ignored  
esp> ␐␂␌␌r�����␌␌␌␌c␌n�|␂�␌�c��on� ...ignored
esp> ␐␂␌␌s�����␌�␌␌c␌n�|␃�␌�p�␌b��on� ...ignored
esp> s��o␌␌�␌l`␂ ...ignored
esp> ready ...matched      
esp> ATE0 ...sent
esp> ATE0 ...ignored
esp> OK ...matched
esp> AT+CIPMUX=1 ...sent
esp> OK ...matched
esp> AT+CIPRECVMODE=1 ...sent
esp> OK ...matched
esp> AT+SYSSTORE=0 ...sent
esp> OK ...matched
esp> AT+CWMODE? ...sent
esp> +CWMODE:1 ...matched
esp> OK ...matched
esp INFO: join AP AP_NAME current
esp> AT+CWJAP="AP_NAME","***" ...sent
esp> ?
esp> busy p... ...ignored
esp> WIFI CONNECTED ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> ?
esp> busy p... ...ignored
esp> WIFI DISCONNECT ...ignored
esp> +CWJAP:4 ...ignored
esp> ERROR ...error
esp ERROR: expected OK got ERROR
esp INFO: quit AP  current
esp> AT+CIPDNS=0 ...sent
esp> OK ...matched
esp> AT+CWDHCP=1,3 ...sent
esp> OK ...matched
esp> AT+CWQAP ...sent
esp> OK ...matched

Connection to WiFi network failed.
```

This line is interesting: ```esp> WIFI CONNECTED ...ignored```
Shouldn't I see "matched" instead?

jandrassy, 2021-05-07:
it can't get an IP address. it could be a DHCP problem, weak signal or insufficient powering

NorbertSandor, 2021-05-07:
> it can't get an IP address. it could be a DHCP problem, weak signal or insufficient powering.

It is 1 metre away from the router, and it connects very quickly when using direct AT commands (communicating directly with the ESP8266).

Besides, if I try the ScanNetworksAdvanced example, it lists the network successfully (it is 1) in the list):
```
Scanning available networks...
** Scan Networks **
number of available networks: 4
1) Signal: -60 dBm	Channel: 1		BSSID: AA:33:C6:36:1B:11
	Encryption: WPA2		SSID: **AP_NAME**
2) Signal: -63 dBm	Channel: 1		BSSID: AC:84:C6:33:11:57
	Encryption: WPA2		SSID: x
3) Signal: -80 dBm	Channel: 1		BSSID: C0:4A:33:11:5C:05
	Encryption: WPA2		SSID: x
4) Signal: -84 dBm	Channel: 8		BSSID: 38:43:11:33:E4:CB
	Encryption: WPA2		SSID: x
```

> the driver waits for OK after CWJAP so yes WIFI CONNECTED is not processed

Besides, now manual AT commands give me this result:
```
ERROR
AT+CWJAP="AP_NAME","***"
WIFI CONNECTED
WIFI DISCONNECT
+CWJAP:4
```

NorbertSandor, 2021-05-07:
Yes, but investigating the cause of this problem is beyond my level of expertice and resources...
Thanks for the lots of help, in the short term I will use the JB AT 1.7 firmware, it seems to be sufficient for my needs (= POSTing to a HTTPS site).

NorbertSandor, 2021-05-07:
> I will use the JB AT 1.7 firmware ..., it seems to be sufficient for my needs (= POSTing to a HTTPS site)

Unfortunatelly it doesn't seem to be stable :(
After some POSTs it won't connect to the server even with plain HTTP.
So I had to downgrade to the original "hated" solution: official 1.7 firmware + plain HTTP connections.
At least it works reliably :\

closed 2021-05-07 by NorbertSandor

jandrassy, 2021-05-08:
the esp8266 has WiFi radio calibration with calibration data stored in flash. but the stored value is not on the same location or is not in the same range for different SDK. so it is necessary to delete it if SDK version is changed. all the firmware you try use different SDK. so it is necessary to erase the flash completely to delete the calibration data. and then the esp8266 should be at a good location until the esp8266 calibrates. 1 m from router is not ideal. the calibration is done at boot. not at every boot. there are some counters for it.

NorbertSandor, 2021-05-08:
> the esp8266 has WiFi radio calibration with calibration data stored in flash. but the stored value is not on the same location or is not in the same range for different SDK. so it is necessary to [...]

I will try it, thanks for the additional info.