JAndrassy/WiFiEspAT #49: Disable AP mode

gimenezfrg, 2021-06-23:
Hi, I'm using this library to connect over a wifi network and send data to an API, everything works fine.

The question is that I realized that ESP is active as if it were a router too, that is, if I look at the available wireless networks the ESP appears, I would like to disable it, is there how?

jandrassy, 2021-06-23:
see the Tools section of WiFiEspAT examples in Arduino IDE Examples menu

gimenezfrg, 2021-06-23:
I can connect and use normally, I just wish it didn't appear as a network, I would like to disable this function if possible.

[image]

[image]

[image]

gimenezfrg, 2021-06-23:
I think I understand, I put these three lines before BEGIN:

```
  WiFi.disconnect(); // to clear the way. not persistent
  WiFi.setPersistent(); // set the following WiFi connection as persistent
  WiFi.endAP(); // to disable default automatic start of persistent AP at startup
```

Is correct?

jandrassy, 2021-06-23:
I don't remember the details, but here is the doc:
https://github.com/jandrassy/WiFiEspAT#the-wifi-object-differences

closed 2021-07-13 by jandrassy