JAndrassy/WiFiEspAT #51: client.connect stop working after some hours

WerkAG, 2021-07-09:
I'm facing a very similar problem than described on here: https://github.com/jandrassy/WiFiEspAT/issues/1
With the required changes, I ported a code which uses WiFiEsp library, to your WiFiEspAT library.
The code sends data to four different servers at different time intervals. Everything works well, but after some hours (usually between 12 to 18) client.connect stops working, and every attempt to connect will fail, even to a server on the local network.
When this happens, I can ping the device running the code, and even access the web server which serves me a little web page. So WiFi is working.
I don't think this problem is caused by an AT 1.7.4 firmware error, because it not happens with the "old" WiFiEsp library, even running with AT 1.7.4  
Your library is a very interesting and useful library, maybe you would like to check and / or try to reproduce the reported issue, because it really happens.
My next step will be add a counter to every client.connect event to try to verify, if the error happens after client.connect is called 255 times, as reported by bialabs, here https://github.com/jandrassy/WiFiEspAT/issues/1

jandrassy, 2021-07-09:
does your sketch use String?

WerkAG, 2021-07-09:
The sketch doesn't freeze. 
As said just any connection, started by client.connect() will fail. There is Wifi connection, and the web server implemented on the sketch (according to debug is using linkId 1) keeps working normally. Its the client.connect() that always fails, and never recover.
The software where I'm trying to use your WiFiEspAT library is a well tested software, which runs 24/7 for many years with the "old" WiFiEsp library.  

For more than 2 months that I'm (not only me, but other members of my development team too) trying to solve this issue when using the WiFiEspAT, without success. That's why we finally decided to report it.

jandrassy, 2021-07-09:
the difference between WiFiEsp and WiFiEspAT library is the  use of AT firmware passive mode for TCP by WiFiEspAT. There could be a bug in AT firmware in passive mode support.

WerkAG, 2021-07-09:
Yes, it could be a problem related to the use the passive mode. I will continue investigating.
I reported it, for you to know that the issue exists.
I tried switch to your library mainly because of two functions not implemented on the original WiFiEsp library, setDNS and hostByname, but I ended up implementing them on the WiFiEsp library, so I can continue using it.
Thank you, and congratulations for your work.

bobemoe, 2021-11-03:
I'm also seeing something very similar to this, if not exactly the same.

I added the counter and it is always exactly at 256 `client.connect()` it stops working. There is a freeze up of about 60 secs (timeout?) and then after this `WiFi.status()` returns `WL_NO_MODULE`.

I'm not running a `WiFiServer` so I cant confirm that that is still responsive.

I feel it gradually gets slower each loop building up to the 256 but I will need to add a time counter to confirm this.

The difference with me to this issue is that I have not ported anything from `WiFiEsp` or know what that is.

I'm using WiFiEspAT 1.3.1 on STM32F4 BlackPill, connected to a ESP8266 running the AT_LoBo firmware.

I wonder if I'm not closing the connection properly? I use `WiFiClient` to make an HTTP POST, read some of the response but call `client.stop()` before the response is finished. I tried reading the remaining response but nothing is `.available()`

I always call `client.stop()` before opening a connection. Sometimes it may not be needed, could calling that too much be an issue?

Might try calling `reset()` after 250 loops, seems like a workaround I'd like to avoid :/

Any thoughts?

@WerkAG did you get any further?

jandrassy, 2021-11-04:
@bobemoe, if you can, try AT 1.7.4

bobemoe, 2021-11-04:
@jandrassy Bingo! Using the official firmware 1.7.4 it has passed loop 256! :) Sounds like this issue needs to be reported to https://github.com/loboris/ESP8266_AT_LoBo

bobemoe, 2021-11-04:
Maybe I spoke too soon.  It just happened again. :(  This time at loop 1020. Seems repeatable.  The number is not as significant as 256, but is close to 1024!

jandrassy, 2021-11-05:
the Adafruit atwinc1500 WiFi shield or module with the Arduino WiFi101 library is much better than the esp8266. the WiFi101 library API is the same as with WiFiEspAT library so it is easy to switch

jandrassy, 2021-12-30:
AT 1.7.5 was published recently as part of SDK 3.0.5
https://github.com/espressif/ESP8266_NONOS_SDK/releases

closed 2022-07-21 by JAndrassy