JAndrassy/WiFiEspAT #53: Issues with receiving status on sending the "AT" command

TinManAkshay, 2021-07-14:
Hi,

This is Akshay again. You have been helping me on Arduino forum. I've looked at these libraries and they are very well structured. I am glad to find these libraries. 

As you mentioned if we want to use Hardware Serial, we can use example "Serialpassthrough". I made a few changes in this example according to the pin out in my board (having ublox NINA W152 and SAMD51 MCU). As per your instructions, I try to send AT on the Serial monitor and it should receive "OK", but its not doing that. 

RX and TX - Pins 19 and 20--------SERCOM0

I am still facing issues with a proper communication with the WiFi module. I cannot figure out what's been going wrong. Is it a coding issue or hardware issue?

Here is my code:

/*
 SerialPassthrough sketch
 with SoftwareSeral option

 created in August 2019 for WiFiEspAT library
 by Juraj Andrassy https://github.com/jandrassy
 */

//#define SAMD_FLOW_CONTROL

// Emulate Serial1 on pins 6/7 if not present
#if defined(ARDUINO_ARCH_AVR) && !defined(HAVE_HWSERIAL1)
#include "SoftwareSerial.h"
SoftwareSerial SerialAT(19, 20); // RX, TX
#define AT_BAUD_RATE 9600
#elif defined(ARDUINO_ARCH_SAMD) && defined(SAMD_FLOW_CONTROL)
#include "wiring_private.h"
Uart SerialAT(&sercom0, 19, 20, SERCOM_RX_PAD_1, UART_TX_PAD_0, 2, 255);
#define AT_BAUD_RATE 115200
#else
#define SerialAT Serial1
#define AT_BAUD_RATE 9600
#endif

void setup() {

  Serial.begin(115200);
  while (!Serial);

  pinMode(PIN_LORA_PWREN, OUTPUT);
  pinMode(PIN_WIFI_PWREN, OUTPUT);
  digitalWrite(PIN_LORA_PWREN, LOW); //turn off the power to LoRa Module...
  digitalWrite(PIN_WIFI_PWREN, HIGH); //turn onn the power to WiFi module...
  
  SerialAT.begin(AT_BAUD_RATE);
#if defined(ARDUINO_ARCH_SAMD) && defined(SAMD_FLOW_CONTROL)
  pinPeripheral(19, PIO_SERCOM_ALT);
  pinPeripheral(20, PIO_SERCOM_ALT);
#endif
}

void loop() {
  while (Serial.available()) {
    SerialAT.write(Serial.read());
  }
  while (SerialAT.available()) {
    Serial.write(SerialAT.read());
  }
}

#if defined(ARDUINO_ARCH_SAMD) && defined(SAMD_FLOW_CONTROL)
void SERCOM0_Handler() {
  SerialAT.IrqHandler();
}
#endif

Thanks,
Akshay

jandrassy, 2021-07-15:
the SERCOM section is not applied in your code because SAMD_FLOW_CONTROL is not defined. the define is commented out

TinManAkshay, 2021-07-15:
I did uncomment out the define statement. Still no luck with receiving the status. Can you please explain how the statements in the loop() are supposed to work?

How are you printing the status on the Serial monitor after sending "AT" command through Serial monitor?

TinManAkshay, 2021-07-15:
This is the output I am getting on the Serial monitor after I run "SerialPassthrough" example:

[image]

jandrassy, 2021-07-16:
SerialPassthrough is a 'tool' sketch. here is the description of the basic SerialPassthrough sketch: https://www.arduino.cc/en/Tutorial/BuiltInExamples/SerialPassthrough.

the SerialPassthrough example in this library is at the same time an UART flow control setup example for SAMD, 
so not a "if you want to use hardware serial", but a if you want to use "hardware serial flow control".
for hardware serial without flow control simply use Serial1.

TinManAkshay, 2021-07-16:
Hi @jandrassy,

Code below (which is similar to your SerialPassthrough) is working:

/* This code is an example of how to use the AT command library to
 *  send AT commands (or ASCII commands) to a module via UART (TX/RX)
 *  using software or hardware serial. The commands can be set up to
 *  check for a specified reply from the module as well as a timeout.
 */

#include "ATcommands.h"
#include <SPI.h>
#include <Wire.h>
//#define SAMD_FLOW_CONTROL
//#if defined(ARDUINO_ARCH_SAMD) && defined(SAMD_FLOW_CONTROL)
#include "wiring_private.h"
Uart SerialAT( &sercom0, PIN_SERIAL1_RX, PIN_SERIAL1_TX, PAD_SERIAL1_RX, PAD_SERIAL1_TX, 2, 255) ;
#define SerialAT Serial1
//#endif

void setup() {
  Serial.begin(9600);
  while (!Serial);
    // Serial Port Hardware Serial1
  SerialAT.begin(115200);
  while(!SerialAT)
  
  delay(1000);
  pinMode(PIN_LORA_PWREN, OUTPUT);
  pinMode(PIN_WIFI_PWREN, OUTPUT);
  pinMode(PIN_WIFI_RESET, OUTPUT);
  pinMode(PIN_WIFI_CTS, OUTPUT);
  digitalWrite(PIN_LORA_PWREN, LOW); //turn off the power to LoRa Module...
  digitalWrite(PIN_WIFI_PWREN, HIGH); //turn onn the power to WiFi module...

//#if defined(ARDUINO_ARCH_SAMD) || defined(SAMD_FLOW_CONTROL)
  pinPeripheral(19, PIO_SERCOM_ALT);
  pinPeripheral(20, PIO_SERCOM_ALT);
//#endif 
  
  WifiReset(LOW, 10); //Resets the pin to LOW for 10ms...

  Serial.println("Enter AT commands");
}
String command, response;

void loop() {
  
  if (Serial.available()) {
    Serial.println("Sending the AT command to NINA");
    command = Serial.readStringUntil('\n'); // Enable newline in serial monitor
    Serial.print("--> "); Serial.println(command); // Debug
    SerialAT.print(command); // Send the command to module    
  }

     // Check for response from module
  if (SerialAT.available()) {
    response = SerialAT.readStringUntil('\n');
    Serial.print("<-- "); Serial.println(response); // Debug
    Serial.println("Response from NINA received SUCCESS");
  }
}

void WifiReset(uint8_t pulse, uint16_t duration){
    if (pulse == LOW) {
    digitalWrite(PIN_WIFI_RESET, HIGH);
    delay(10);
    digitalWrite(PIN_WIFI_RESET, LOW);
    delay(duration);
    digitalWrite(PIN_WIFI_RESET, HIGH);
  }
  else if (pulse == HIGH) {
    digitalWrite(PIN_WIFI_RESET, LOW);
    delay(10);
    digitalWrite(PIN_WIFI_RESET, HIGH);
    delay(duration);
    digitalWrite(PIN_WIFI_RESET, LOW);
  }
}

//#if defined(ARDUINO_ARCH_SAMD) || defined(SAMD_FLOW_CONTROL)
void SERCOM0_Handler() {
  SerialAT.IrqHandler();
}
//#endif

[image]

Till now, I am able to see the correct status after sending the commands like AT, AT+CGMR, AT+CGMM, AT+UWSCAN, AT+UWSC and so on.

Now I am trying to test your "SetupPersistentWifiConnection" to connect to the Wifi. I've made few changes in this example according to my code above. I cannot talk to the AT firmware and cant connect to the Wifi. 

Could you please guide how I should proceed with example, SetupPersistentWifiConnection?

jandrassy, 2021-07-16:
> Could you please guide how I should proceed with example, SetupPersistentWifiConnection?

open the example, switch to arduino-secrets tab, fill in ssid and password and hit Upload. then open Serial Monitor and see the log

TinManAkshay, 2021-07-16:
I did change the network details in arduino secrets.h. Serial monitor is showing this:

[image]

jandrassy, 2021-07-17:
so your Serial1 doesn't connect to esp32?
then use the `Uart SerialAT( &sercom0, ` UART as in the working sketch

TinManAkshay, 2021-07-19:
I did that too. But its not talking to the module. Same error as above. I believe the commands you used in your libraries are not in the datasheet (for ublox nina w152). Every command starts with AT+UWxxxx, for an instance, UWSCAN, UWSC, UWSCA, UWCGM and so on.

Thoughts? Feedback?

jandrassy, 2021-07-20:
my library is for Espressif's esp32 AT firmware

TinManAkshay, 2021-07-20:
I understand what you are saying. I've observed that AT command set structure are different both for Espressif's esp32 (https://docs.espressif.com/projects/esp-at/en/latest/AT_Command_Set/) and NINA W152 (acc. to the datasheet). 

If I run your example sketches through Espressif's esp32 AT firmware, will it work on NINA module even though there is a difference in a AT command sets?

jandrassy, 2021-07-20:
> If I run your example sketches through Espressif's esp32 AT firmware, will it work on NINA module even though there is a difference in a AT command sets?
how could it work?

TinManAkshay, 2021-07-20:
As you mentioned AT firmware 1.7 is only for ESP8826 and I am asking if I flash AT firmware 2.0 or higher into my NINA W152 (uses ESP32-D0WDQ6) module using esp32tools, then I will be able to run your example sketches?

TinManAkshay, 2021-07-20:
I've been trying to flash AT firmware 2.0 and higher and here is the error shows up while flashing the firmware:

[image]

closed 2021-12-30 by jandrassy