JAndrassy/WiFiEspAT #54: ConfigurationAP example not working

gcharles81, 2021-07-15:
Hello I am using this library with Arduino Due and it works great I have UDP connection and also webserver running but recently I started to experiment with the ConfigurationAP example and this example is not working for me also without any code change the programm stops exactly after "Connection successful. Ending AP." and programm is never restarted and wifi module never connects again to the received SSID and Password from the webserver page

Thanks

jandrassy, 2021-07-15:
did you remove the `WiFi.disconnect();` line?

gcharles81, 2021-07-15:
No I haven't removed it

gcharles81, 2021-07-15:
Same issue if I remove WiFi.disconnect();

Serial log below 

19:13:54.862 -> Could not connect to WiFi. Starting configuration AP...
19:13:54.862 -> Connect your computer to the WiFi network  NN002
19:13:54.862 -> and enter http://192.168.4.1 in a Web browser
19:14:24.021 -> 
19:14:24.021 -> Attempting to connect to WPA SSID: Marsa
19:14:31.616 -> Connection successful. Ending AP.

jandrassy, 2021-07-16:
that is all what the example does

gcharles81, 2021-07-22:
Hello in my case I have the below loop as a test and I am not getting any serial print statements that's why I came to a conclusion that the code is blocked after "Connection successful. Ending AP."

void loop() {
  // put your main code here, to run repeatedly:

  delay(2000);
  Serial.println("BEEP");
}

jandrassy, 2021-07-22:
what AT firmware version do you use?

jandrassy, 2021-12-30:
`return;` was missing after AP finished. fixed in version 1.3.2

closed 2021-12-30 by jandrassy