JAndrassy/WiFiEspAT #55: Problems with SNTPTime example

DavidMR91, 2021-07-17:
Hello, I'm having this problem with the SNTPTime example:

```
Waiting for connection to WiFi
.
Waiting for SNTP
............................................................................................................................................................................................................................
```

jandrassy, 2021-07-18:
as documented in firmware compare table in README and in the comment on top of the example, getTime doesn't work with Espressif AT 1

DavidMR91, 2021-07-18:
My esp8266 have the Espressif AT firmware 2.1 downloaded since https://github.com/jandrassy/UnoWiFiDevEdSerial1/wiki/files/ESP8266-1MB-tx1rx3-AT_V2.1.zip and uploaded with esptool.py

DavidMR91, 2021-07-18:
fixed, I forgot to comment #define WIFIESPAT1 in Documents\Arduino\libraries\WiFiEspAT\src\utility\EspAtDrvTypes.h

closed 2021-07-18 by jandrassy