JAndrassy/WiFiEspAT #56: AT Mega 2.1 Issue

DavidMR91, 2021-07-19:
Hello, 
After downloading https://github.com/jandrassy/UnoWiFiDevEdSerial1/wiki/files/ESP8266-1MB-tx1rx3-AT_V2.1.zip and uploading it to the ESP8266 using esptool.py, appears like v2.2 not v2.1:
[image]

jandrassy, 2021-07-19:
yes. but it is 2.1

closed 2021-07-19 by jandrassy

DavidMR91, 2021-07-19:
Why if I download, compile and upload the V2.1 if I run the firmware version checker, say V2.2.0.0-dev?
I'm trying to downgrade the AT Firmware, and every time apear V2.2.0.0-dev. The firmware is loading correctly, and I tried to upload a blank sketch and then the AT Firmware 2.1 and V2.2.0.0-dev still appears

jandrassy, 2021-07-19:
because I built the 1 MB version after they changed the version number, but before any other changes. with AT+GMR you can see that the build timestamp is from last year