JAndrassy/WiFiEspAT #57: AT Firmware 2.2 OTA problems

DavidMR91, 2021-07-19:
Hello,
The last few weeks I have been playing and working with SerialWiFiOTA with AT Firmware 2.1 on ESP8266 and with an ATMega 2560 with communication between them at 250,000 baud.
With this configuration, everything worked perfect, but when uploading AT Firmware 2.2, it seems that it has stopped working. I attach a photo:
[image]
Do you know what could be happening?

jandrassy, 2021-07-19:
everything else works with AT 2.2? only upload doesn't?

DavidMR91, 2021-07-19:
Right, the rest of things seem to work all

DavidMR91, 2021-07-19:
Sorry, my fault, I changed computers and I forgot a step:
I forgot to copy file "platform.local.txt" from "C:\Users\[USER]\Documents\Arduino\libraries\ArduinoOTA\extras\avr" to "C:\Program Files (x86)\Arduino\hardware\arduino\avr"

closed 2021-07-19 by DavidMR91

jandrassy, 2021-07-19:
now I see that it used the old upload command version