JAndrassy/WiFiEspAT #58: MQTT Connection Skip a second or RC -2 or -3

leejames92, 2021-08-14:
Hi, 

I have this issue of which I am unable to solve thus far. My ESP8266 paired with Arduino Mega and DS3231 RTC is setup to publish data every second to thingsboard IOT online through my home WiFi.

Most times per second works fine, but occasionally it would skip a second and after 12 hours or so (sometimes sooner) it would disconnect (but able to reconnect). I would like to eliminate the skip a second + the MQTT disconnect entirely to be able to say my system is stable and reliable. 

Sample of sketch in Arduino,

```
#include <Wire.h>
#include <LCD.h>
#include <LiquidCrystal_I2C.h>
#include <SoftwareSerial.h>
#include "RTClib.h"
#include <SPI.h>
#include "DHT.h"
#include <WiFiClient.h>
#include <WiFiEspAT.h>

#include <WiFiUdp.h>
#include <PubSubClient.h>
#include "SoftwareSerial.h"

#define WIFI_AP "HomeWifiSSID"
#define WIFI_PASSWORD "HomePassword"

#define TOKEN "ThingsboardAccessToken"

char thingsboardServer[] = "thingsboard.cloud";
WiFiClient espClient;
PubSubClient client(espClient);

int status = WL_IDLE_STATUS;
unsigned long lastSend;

unsigned long previousDT = 0;
int uptime;  
int status2;
int status3;
int status4;
int status5;
int status6;
int status7;
int status8;

void setup()
{
  Serial.begin(38400);
  Serial1.begin(38400);
  pinMode (sensor1, INPUT);
  pinMode (sensor2, INPUT);
  pinMode (sensor3, INPUT);
  pinMode (sensor4, INPUT);
  pinMode (sensor5, INPUT);
  pinMode (sensor6, INPUT);
  pinMode (sensor7, INPUT);
  pinMode (sensor8, INPUT);
  InitWiFi();
  client.setServer( thingsboardServer, 1883 );
  lastSend = 0;
  WiFi.sleepMode(WIFI_NONE_SLEEP);

  if ( client.connect ("ClientName", "James", "Password"))  {
    Serial.println( "[DONE]" );
  }
  boolean uptime = digitalRead(sensor1);
}

void loop()
{
[... 318 lines cut ...]
esp> STATUS:3 ...matched
esp> +CIPSTATUS:4,"TCP","54.85.19.206",1883,12611,0 ...ignored
esp> OK ...matched
WiFi OK **seconds: 81059**esp INFO: send data on link 4                           <<<<<<<<<<<<<<Here for seconds
esp> AT+CIPSEND=4,124 ...sent
esp> OK ...ignored
esp> > ...matched
esp> Recv 124 bytes ...matched
esp> SEND OK ...matched
esp INFO: 	sent 124 bytes on link 4
 {"uptime":1,"status2":1,"status3":1,"status4":1,"status5":1,"status6":1,"status7":1,"status8":1}
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:3 ...matched
esp> +CIPSTATUS:4,"TCP","54.85.19.206",1883,12611,0 ...ignored
esp> OK ...matched
WiFi OK

```

As you can see above **bolded seconds** can see a 1 second skip. I really don't know why is this happening.

Is it because of the baud rate? or a delay in a sketch? I tried many possible solutions but to no avail :( 

Please help thank you.

jandrassy, 2021-08-15:
maybe first fix the presented sketch. variable `seconds` is not declared. global variables are shadowed by local variables...
please edit the original post.

leejames92, 2021-08-16:
Hi jassandry,

Thanks for the find. I've just amended in my code and testing now. 

However, this is an error log I've captured using the original code and the MQTT connection broke,

```
 WiFi state: 0esp INFO: AP query
esp> AT+CWJAP? ...sent
esp> +CWJAP:"HomeWifiSSID","7c:8b:ca:3e:d7:5b",11,-55,0 ...matched
esp> OK ...matched
Signal strength (RSSI):-55 dBm
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:3 ...matched
esp> +CIPSTATUS:4,"TCP","54.85.19.206",1883,12611,0 ...ignored
esp> OK ...matched
esp INFO: sync
esp> AT+CIPSTATUS ...sent
esp> STATUS:3 ...matched
esp> +CIPSTATUS:4,"TCP","54.85.19.206",1883,12611,0 ...matched
esp> OK ...end of list
esp> AT+CIPRECVLEN? ...sent
esp> +CIPRECVLEN:0,0,0,0,0 ...matched
esp> OK ...matched
Machine Status..
Uptime: 1,1,1,1,1,1,1,1,esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:3 ...matched
esp> +CIPSTATUS:4,"TCP","54.85.19.206",1883,12611,0 ...ignored
esp> OK ...matched
WiFi OK seconds: 21421esp INFO: send data on link 4
esp> AT+CIPSEND=4,124 ...sent
esp> OK ...ignored
esp> > ...matched
esp> Recv 124 bytes ...matched
esp> SEND OK ...matched
esp INFO: 	sent 124 bytes on link 4
 {"uptime":1,"status2":1,"status3":1,"status4":1,"status5":1,"status6":1,"status7":1,"status8":1}
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:3 ...matched
esp> +CIPSTATUS:4,"TCP","54.85.19.206",1883,12611,0 ...ignored
esp> OK ...matched
WiFi OK

WiFi state: 0esp INFO: AP query
esp> AT+CWJAP? ...sent
esp> +CWJAP:"HomeWifiSSID","7c:8b:ca:3e:d7:5b",11,-57,0 ...matched
esp> OK ...matched
Signal strength (RSSI):-57 dBm
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:3 ...matched
esp> +CIPSTATUS:4,"TCP","54.85.19.206",1883,12611,0 ...ignored
esp> OK ...matched
esp INFO: sync
esp> AT+CIPSTATUS ...sent
esp> STATUS:3 ...matched
esp> +CIPSTATUS:4,"TCP","54.85.19.206",1883,12611,0 ...matched
esp> OK ...end of list
esp> AT+CIPRECVLEN? ...sent
esp> +CIPRECVLEN:0,0,0,0,0 ...matched
esp> OK ...matched
Machine Status..
Uptime: 1,1,1,1,1,1,1,1,esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:3 ...matched
esp> +CIPSTATUS:4,"TCP","54.85.19.206",1883,12611,0 ...ignored
esp> OK ...matched
WiFi OK seconds: 21422esp INFO: send data on link 4
esp> AT+CIPSEND=4,124 ...sent
esp> OK ...ignored
esp> > ...matched
esp> Recv 124 bytes ...matched
esp> ?
esp> busy s... ...ignored
esp> ?
esp> busy s... ...ignored
esp> ?
esp> busy s... ...ignored
esp> ?
esp> busy s... ...ignored
esp> ?
esp> busy s... ...ignored
esp> ?
esp> busy s... ...ignored
esp> ?
esp> busy s... ...ignored
esp> 4,CLOSED ...processed
esp INFO: closed linkId 4
esp> SEND FAIL ...matched
esp ERROR: failed to send data
 {"uptime":1,"status2":1,"status3":1,"status4":1,"status5":1,"status6":1,"status7":1,"status8":1}
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:4 ...matched
esp> OK ...matched
WiFi OK

 WiFi state: 0esp INFO: AP query
esp> AT+CWJAP? ...sent
esp> +CWJAP:"HomeWifiSSID","7c:8b:ca:3e:d7:5b",11,-56,0 ...matched
esp> OK ...matched
Signal strength (RSSI):-56 dBm
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:4 ...matched
esp> OK ...matched
Machine Status..
Uptime: 1,1,1,1,1,1,1,1,esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:4 ...matched
esp> OK ...matched
WiFi OK seconds: 21429 {"uptime":1,"status2":1,"status3":1,"status4":1,"status5":1,"status6":1,"status7":1,"status8":1}
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:4 ...matched
esp> OK ...matched
WiFi OK

WiFi state: -3esp INFO: AP query
esp> AT+CWJAP? ...sent
esp> +CWJAP:"HomeWifiSSID","7c:8b:ca:3e:d7:5b",11,-57,0 ...matched
esp> OK ...matched
Signal strength (RSSI):-57 dBm
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:4 ...matched
esp> OK ...matched
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:4 ...matched
esp> OK ...matched
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:4 ...matched
esp> OK ...matched
WiFi OK

 WiFi state: -3esp INFO: AP query
esp> AT+CWJAP? ...sent
esp> +CWJAP:"HomeWifiSSID","7c:8b:ca:3e:d7:5b",11,-54,0 ...matched
esp> OK ...matched
Signal strength (RSSI):-54 dBm
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:4 ...matched
esp> OK ...matched
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:4 ...matched
esp> OK ...matched
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:4 ...matched
esp> OK ...matched
WiFi OK
```

I am unable to decipher the issue based on this log. Do hope I can get some clarity and assistance. 
Looks like their is a send fail, then send once ok after skipping couple of seconds, then the MQTT connection broke right after.

jandrassy, 2021-08-16:
AT firmware responded `SEND FAIL`. there is no more info

leejames92, 2021-08-16:
Meaning there is an issue with my sketch? Sorry i'm quite unfamiliar still with this MQTT related matters

jandrassy, 2021-08-16:
> 
> 
> Meaning there is an issue with my sketch? Sorry i'm quite unfamiliar still with this MQTT related matters

I guess not. What version of AT firmware is it?

leejames92, 2021-08-16:
AT 1.7.4.0

Is it esp8266 wifi connection cut off or its busy?

jandrassy, 2021-08-16:
> AT 1.7.4.0
> Is it esp8266 wifi connection cut off or its busy?

1.7.4 is the best ESP AT firmware ever (I doubt AT 2 can be so stable in near or far future), but event then it can have problems for various reasons.
Does the next sending work?

leejames92, 2021-08-16:
I see. I tried again and after 8 hours the MQTT connection broke again :/

```
 WiFi state: 0esp INFO: AP query
esp> AT+CWJAP? ...sent
esp> +CWJAP:"HomeWiFiSSID","7c:8b:ca:3e:d7:5b",1,-68,0 ...matched
esp> OK ...matched
Signal strength (RSSI):-68 dBm
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:3 ...matched
esp> +CIPSTATUS:4,"TCP","54.85.19.206",1883,43552,0 ...ignored
esp> OK ...matched
esp INFO: sync
esp> AT+CIPSTATUS ...sent
esp> STATUS:3 ...matched
esp> +CIPSTATUS:4,"TCP","54.85.19.206",1883,43552,0 ...matched
esp> OK ...end of list
esp> AT+CIPRECVLEN? ...sent
esp> +CIPRECVLEN:0,0,0,0,0 ...matched
esp> OK ...matched
Machine Status..
Uptime: 1,1,1,1,1,1,1,1,esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:3 ...matched
esp> +CIPSTATUS:4,"TCP","54.85.19.206",1883,43552,0 ...ignored
esp> OK ...matched
WiFi OK seconds: 59821esp INFO: send data on link 4
esp> AT+CIPSEND=4,124 ...sent
esp> OK ...ignored
esp> > ...matched
esp> Recv 124 bytes ...matched
esp> SEND OK ...matched
esp INFO: 	sent 124 bytes on link 4
 {"uptime":1,"status2":1,"status3":1,"status4":1,"status5":1,"status6":1,"status7":1,"status8":1}
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:3 ...matched
esp> +CIPSTATUS:4,"TCP","54.85.19.206",1883,43552,0 ...ignored
esp> OK ...matched
WiFi OK

 WiFi state: 0esp INFO: AP query
esp> AT+CWJAP? ...sent
esp> +CWJAP:"HomeWiFiSSID","7c:8b:ca:3e:d7:5b",1,-65,0 ...matched
esp> OK ...matched
Signal strength (RSSI):-65 dBm
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:3 ...matched
esp> +CIPSTATUS:4,"TCP","54.85.19.206",1883,43552,0 ...ignored
esp> OK ...matched
esp INFO: sync
esp> AT+CIPSTATUS ...sent
esp> STATUS:3 ...matched
esp> +CIPSTATUS:4,"TCP","54.85.19.206",1883,43552,0 ...matched
esp> OK ...end of list
esp> AT+CIPRECVLEN? ...sent
esp> +CIPRECVLEN:0,0,0,0,0 ...matched
esp> OK ...matched
Machine Status..
Uptime: 1,1,1,1,1,1,1,1,esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:3 ...matched
esp> +CIPSTATUS:4,"TCP","54.85.19.206",1883,43552,0 ...ignored
esp> OK ...matched
WiFi OK seconds: 59822esp INFO: send data on link 4
esp> AT+CIPSEND=4,124 ...sent
esp> OK ...ignored
esp> > ...matched
esp> Recv 124 bytes ...matched
esp> ?
esp> busy s... ...ignored
esp> ?
esp> busy s... ...ignored
esp> ?
esp> busy s... ...ignored
esp> ?
esp> busy s... ...ignored
esp> ?
esp> busy s... ...ignored
esp> ?
esp> busy s... ...ignored
esp> ?
esp> busy s... ...ignored
esp> 4,CLOSED ...processed
esp INFO: closed linkId 4
esp> SEND FAIL ...matched
esp ERROR: failed to send data
 {"uptime":1,"status2":1,"status3":1,"status4":1,"status5":1,"status6":1,"status7":1,"status8":1}
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:4 ...matched
esp> OK ...matched
WiFi OK

 WiFi state: 0esp INFO: AP query
esp> AT+CWJAP? ...sent
esp> +CWJAP:"HomeWiFiSSID","7c:8b:ca:3e:d7:5b",1,-63,0 ...matched
esp> OK ...matched
Signal strength (RSSI):-63 dBm
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:4 ...matched
esp> OK ...matched
Machine Status..
Uptime: 1,1,1,1,1,1,1,1,esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:4 ...matched
esp> OK ...matched
WiFi OK seconds: 59830 {"uptime":1,"status2":1,"status3":1,"status4":1,"status5":1,"status6":1,"status7":1,"status8":1}
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:4 ...matched
esp> OK ...matched
WiFi OK

 WiFi state: -3esp INFO: AP query
esp> AT+CWJAP? ...sent
esp> +CWJAP:"HomeWiFiSSID","7c:8b:ca:3e:d7:5b",1,-64,0 ...matched
esp> OK ...matched
Signal strength (RSSI):-64 dBm
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:4 ...matched
esp> OK ...matched
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:4 ...matched
esp> OK ...matched
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:4 ...matched
esp> OK ...matched
WiFi OK

 WiFi state: -3esp INFO: AP query
esp> AT+CWJAP? ...sent
esp> +CWJAP:"HomeWiFiSSID","7c:8b:ca:3e:d7:5b",1,-62,0 ...matched
esp> OK ...matched
Signal strength (RSSI):-62 dBm
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:4 ...matched
esp> OK ...matched
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:4 ...matched
esp> OK ...matched
esp INFO: wifi status
esp> AT+CIPSTATUS ...sent
esp> STATUS:4 ...matched
esp> OK ...matched
WiFi OK

```

Again it is seen that the esp is busy first. and then the connection just break :(

jandrassy, 2021-08-16:
Does the next sending work?

leejames92, 2021-08-16:
Sorry, what do you mean by next sending? Is it restart my arduino mega and esp8266? if that, yes it work. 

Also if I include reconnect after MQTT connection loss, that'll work as well I believe.

leejames92, 2021-08-16:
Maybe I think you meant my client.state()? It becomes - 3. 

When it becomes -3 then it loops without sending anymore

jandrassy, 2021-08-16:
```
esp> AT+CIPSTATUS ...sent
esp> STATUS:4 ...matched
```
status 4 is "The TCP or UDP transmission of ESP8266 Station is disconnected"
which is expected, because the TCP connection closing was announced by AT firmware here
```
esp> busy s... ...ignored
esp> 4,CLOSED ...processed
```
but there is no new connection attempt. I don't know what code you run.
**maybe the cloud kicks you out for too frequent calls.**

leejames92, 2021-08-17:
I see. I'm disappointed with self unable to solve this issue. 

However I bought a WeMos arduino mega with installed on board esp8266 and hopefully it'll work when it arrives. Maybe my current setup esp8266 isn't getting enough power.

closed 2021-12-30 by jandrassy