JAndrassy/WiFiEspAT #68: setHostname doesn't work if the ESP is set in AP mode

rsteenbu, 2021-12-05:
Hi Jaraj,

I've been fighting with setHostname for a few days, it's been returning error.  Here's the debug output:
esp INFO: soft reset
esp INFO: wifi status
esp INFO: set hostname teensytest
esp ERROR: expected OK got ERROR
esp INFO: join AP ######### current
esp INFO: hostname query
Hostname set to: ESP-36CA86
Waiting for connection to WiFi
esp INFO: wifi status

After that it goes on and connects to my wifi just fine. 

According to the ESP8266 documentation, setting the hostname via AT command will return error if the ESP is set in AP mode.  I had assumed that was the problem, so I added the code to EspAtDrvClass::setHostname method  to set mode to STA mode.

It works now, but I'd like to know if I'm missing something.  I think it's kind of weird that no-one else seems to have it this issue that I can find.   Because the ESP needs to be in STA mode for setting the hostname to work it seems like a good change nonetheless.

Regard,

Richard

jandrassy, 2021-12-05:
why is your esp8266 not in station mode at boot?

jandrassy, 2021-12-05:
yes setHostname should turn on the station mode as config and setDNS do.

rsteenbu, 2021-12-07:
Thanks, I just submitted a PR for the same.

rsteenbu, 2021-12-07:
Thanks for the merge!