JAndrassy/WiFiEspAT #70: [question] which ssid is my esp currently connected?

atesin, 2021-12-19:
hi.....  i have some esp8266 with join remote ap persistent on its own flash, not set by me

is there any way to find which ssid are currently connected??... like the `WiFi.localIP()` function that tells the station ip assigned by remote ap, but a command to tell  the actual name of external ap.... i have many wifis surrounding, i need this function to know which wifi connect my devices to reach the esp (imagine ssid and ip will be printed on an lcd)

... i think the low level command could be `AT+CWJAP_CUR?` but i am not sure if this command tells you the remote ssid value set in ram with command or the realtime connection to ap

i couldn't found any reference on documentation

atesin, 2021-12-19:
i answer myself... accordig the `PrintPersistentSettings,ino` example, the function seems to be `WiFi.SSID(charBuffer)` which puts the currently connected ap ssid in char buffer...  this differs a little from wifinina api, this only works on curently connected wifi ans is reentrant

not tested yet... any additional comment? ..  thanks

jandrassy, 2021-12-19:
https://github.com/jandrassy/WiFiEspAT#enhanced-wifi-api

> SSID optionally can be called with char array to fill (see PrintPersistentSettings.ino tool example)

the WiFiNINA way works too

closed 2021-12-30 by jandrassy