JAndrassy/WiFiEspAT #71: [suggestion] new function WiFiClient.localPort()

atesin, 2022-01-04:
the library currently has a WiFiClient.remotePort() function, but have no WiFiClient.localPort() one...  it could be useful, imagine this situation...

we start a server, accept() incoming connections and remember them some way for future use... we can distinguish them by remote ip/port and do something customized action with each them... for example have another lists with connection status, or total data read, or time connected, etc...  for this we can distinguish connections by ip/port for example (i.e. socket: srcIp:srcPort:dstIp:dstPort)

suppose at some time we need to retrieve simultaneously some data from another server to continue, still keeping connections sorted and managed....    but the remote port will be always fixed, meaning if we open 2 connections we will have no way to dstinguish them... in this case (as client) we could distinguish them by local port since it will be random, but currently there is no way to get it :(

jandrassy, 2022-01-04:
done

closed 2022-01-04 by jandrassy

jandrassy, 2022-01-04:
note that a connection closed in AT firmware may be still open in library if it has cached data not read by the sketch. then the remoteIP, remotePort and localPort will return zeros.

atesin, 2022-01-04:
excelent!...  so the 0-value is the "standard" behavior for closed connections parameters...  could even be useful to identify early those with unread data

keep in mind this new function is not present in wifinina api, so you will have to put in documentation also

(the best esp8266 arduino lib!)

jandrassy, 2022-01-04:
```
uint8_t WiFiClient::status() {
  if (!stream)
    return CLOSED;
  if (stream->connected())
    return ESTABLISHED;
  return CLOSED;
}
```

atesin, 2022-04-21:
just as a complement, imagine this other inverse situation...

- you start a connection as a client to a remote server, let's say a telnet server with known port 23
- `WiFiClient.remotePort()` will always return `23`
- but you have no idea which local port the esp8266 had opened, so you can't identify/distinguish the connection

jandrassy, 2022-04-21:
I closed this issue when I added WiFiClient.localPort(). what is the problem?