JAndrassy/WiFiEspAT #77: cannot set softap persistent

atesin, 2022-04-27:
hi... in my sketch i have these commands (i previously configured ap parameters)
```
  WiFi.setPersistent();
  WiFi.beginAP();
```
everything is fine, i can detect the wifi with my phone and can connect with my parameters... then as the configurarion is stored in the esp i commented these lines and reflash the sketch

but if i hard-reset the arduino+esp reconnecting power the wifi was gone... so i repeat the process with logging enabled and that is what i saw
```
esp INFO: begin SoftAP persistent
esp> AT+CWMODE_CUR=3 ...sent
esp> OK ...matched
```
i believe for persistent mode the command should be `AT+CWMODE_DEF=3` instead the `_CUR`one ... furthermore i grepped the `_DEF` string in all files and just found 2 lines in `EspAtDrv.cpp` referencing `AT+CIPDNS_DEF` command, nothing more...  there are many persistent commands ended with `_DEF` present in my firmware, it is based on espressif nonos sdk ver 3.0.1

[lobo AT reference documentation](https://github.com/loboris/ESP8266_AT_LoBo/blob/master/at_lobo/doc/4a-esp8266_at_instruction_set_en.pdf)

i have LoBo firmware and esp-01s (the tiny black 2x4 pins 2 chips esp with 1M flash)
[image]

JAndrassy, 2022-04-27:
without _DEF it is same as _DEF in AT 1.
I will check the softAP persistence

JAndrassy, 2022-04-28:
SoftAP mode is not made persistent. the SoftAP settings are persistent

atesin, 2022-04-28:
> SoftAP mode is not made persistent. the SoftAP settings are persistent. you can use WiFi.beginAP() without parameters to start the SoftAP with persistent settings.

and how about `AT+CWMODE_DEF=3` or 2 ??

[image]
[image]

[image]
[image]
[image]

JAndrassy, 2022-04-29:
beginAP doesn't change remembered CWMODE. I decided that when I was creating the library.

closed 2022-04-29 by JAndrassy

atesin, 2022-04-29:
sorry for post before closing .....  that means that if i set `AT+CWMODE_DEF` externally (with an ftdi and AT commands for example), this lib will never overwrite that configuration *stored on the flash* ??  
good to know it

JAndrassy, 2022-04-29:
> sorry for post before closing ..... that means that if i set `AT+CWMODE_DEF` externally (with an ftdi and AT commands for example), this lib will never overwrite that configuration _stored on the flash_ ?? good to know it

I think, it will overwrite CWMODE_DEF if WiFi.begin is called with persistent true.