JAndrassy/WiFiEspAT #79: quering the wifi mode (station, softap, both)

atesin, 2022-04-30:
hi... i need to know if my esp is working on softap (not exclusive)... my first instinct tell me to use the `status()`command  
- the [Enhanced WiFi API](https://github.com/jandrassy/WiFiEspAT#enhanced-wifi-api) section tells me this lib is used similar to [WiFiNINA library](https://www.arduino.cc/en/Reference/WiFiNINA)
- [WiFi.status() documentation](https://www.arduino.cc/reference/en/libraries/wifinina/wifi.status/) shows many return states, i am interested in `WL_AP_LISTENING` or `WL_AP_CONNECTED`
- looking the source code i noticed `WL_AP_LISTENING` is just used once in `WiFiClass::beginAP()`, while `WL_AP_CONNECTED`isn't even mentioned
- so looking for some `AT+CWMODE_CUR?`/`AT+CWMODE?` command wrapper, the only i saw is in `EspAtDrvClass::reset()` function

so i think i have currently no way to know if my softap is running or not (maybe the same with station mode)

i noticed the `status()` function is mainly targeted to station mode... talking about WiFiNINA api, i dont know what status it would return me if i have both station+softap modes enabled, is ambiguous... or both modes running but no clients connected, or both modes running but one with clients and the other no... maybe should have 2 statuses, one for station and the other for softap

atesin, 2022-04-30:
i posted a [similar question](https://github.com/arduino-libraries/WiFiNINA/issues/221) in WiFiNINA repo page

JAndrassy, 2022-04-30:
as the library doesn't set SoftAP persistent start, the AP only runs if the sketch started it.
the same applies for WiFiNina.

atesin, 2022-05-01:
if the esp has commands like AT+CWMODE_DEF and AT+AUTOCONN to start (softap mode / station mode / autoconnect ) ... which do you think is the criteria in WiFiEspAT and WiFiNINA to **NOT** set these modes/connections start at boot?? ...  this way who need are forced to plug an ftdi and send CWMODE_DEF by hand as the libs dont set it

if somebody dont want the mode to be saved just dont use setPersistent()  ...  as currently setPersistent() dont saves all the modes so is not fully functional, in my humble point of view ... could you clarify this a little more? .. thanks

JAndrassy, 2022-05-01:
SetupPersistentWiFiConnection has
```
  WiFi.setPersistent();
  WiFi.endAP(); // to disable default automatic start of persistent AP at startup
```
endAP(true) or endAP() with setPersistent disable automatic start of SoftAP

atesin, 2022-05-01:
maybe i didn't explain myself enough, this is my situation, maybe somebody else have a similar one

i am developing a system that works together between arduino and my server (http + php + mariadb), a low cost irrigation system, but also could be a domotics or an automation one

when the farmer receives the device (arduino + esp8266) he must provision his wifi for the system starts to run.... then the wifi access should be saved and started automatically, and the farmer (often not so skilled in technology) should NOT enter wifi credentials everytime the system shuts down (by energy failures, signal lost or whatever)

so my question actually is... how can i make the esp8266 save wifi credentials and CONNECT AUTOMATICALLY when boot using this library... because i don't understand why it wasn't designed to auto-connect, a flaw imho

JAndrassy, 2022-05-01:
https://github.com/khoih-prog/WiFiManager_Generic_Lite
or the ConfigurationAP example

atesin, 2022-05-02:
> The AVR-family boards (UNO, Nano, Mega, etc.) are not supported

- with this library, how can i know if SoftAP is running or not (with or without clients)?
- why WiFiEspAT and WiFiNINA libraries can't start a SoftAP automatically when boot?

JAndrassy, 2022-05-02:
>     * with this library, how can i know if SoftAP mode is active or not (with or without clients)?

you can't. but you can stop it and then it is not running or start it and then it is running. STA connection can drop so we need a status() function. but we have full control over SoftAP.
 
>     * why WiFiEspAT and WiFiNINA libraries can't set a SoftAP to start automatically when boot?

the STA connection takes time so it is better for WiFiEspAT to let the esp start it as soon as possible and do it asynchronous to setup(), while SoftAP starts almost immediately.