JAndrassy/WiFiEspAT #98: const OK and STATUS redeclared

Playit3110, 2023-01-13:
```
C:\...\WiFiEspAT\src\utility\EspAtDrv.cpp:37:15: error: 'const char OK []' redeclared as different kind of symbol
 const char OK[] PROGMEM = "OK";
               ^
...

C:\...\esp32\1.0.6/tools/sdk/include/esp32/rom/ets_sys.h:630:5: note: previous declaration 'STATUS OK'
     OK = 0,
     ^
...

C:\...\WiFiEspAT\src\utility\EspAtDrv.cpp:38:19: error: 'const char STATUS []' redeclared as different kind of symbol
 const char STATUS[] PROGMEM = "STATUS";
                   ^

...
C:\...\esp32\1.0.6/tools/sdk/include/esp32/rom/ets_sys.h:635:3: note: previous declaration 'typedef enum STATUS STATUS'
 } STATUS;
   ^
```

__Version__
WiFiEspAT: 1.4.1
ESP32 tools: 1.0.6

And now i cant compile the code and upload it. How can I fix it?

JAndrassy, 2023-01-13:
what do you want to do with this library on esp32?

closed 2023-01-13 by JAndrassy

Playit3110, 2023-01-13:
i want to connect to a wifi AP, but i have two APs with the same SSID and need to also set the BSSID, which is only possible with WiFiEspAT.

JAndrassy, 2023-01-13:
do you have an esp8266 or esp32 with AT firmware connected to your esp32?

Playit3110, 2023-01-13:
ESP32. Specificly ESP32-CAM

JAndrassy, 2023-01-13:
read my question again

Playit3110, 2023-01-13:
I have an UART Serial connection to the ESP32 but im not sure what you try to ask for

JAndrassy, 2023-01-13:
do you have two esp32 connected over Serial? one with AT firmware and second as host with this library?

Playit3110, 2023-01-13:
Computer -- USB-to-Serial -- ESP32-CAM

JAndrassy, 2023-01-13:
the WiFiEspAT library is not for the esp32

Playit3110, 2023-01-13:
so i cant use it with the esp32? how do i know connect to my network without changing the SSID?

Playit3110, 2023-01-14:
So this does not apply anymore?

https://forum.arduino.cc/t/connecting-wifi-to-specific-mac-address/617357/3

JAndrassy, 2023-01-14:
> So this does not apply anymore?
> 
> https://forum.arduino.cc/t/connecting-wifi-to-specific-mac-address/617357/3

that was not about this library

Playit3110, 2023-01-14:
Please help clear it up. What is it about?

JAndrassy, 2023-01-14:
> Please help clear it up. What is it about?

so OP didn't specify what WiFi library they use, so I listed all, starting with link to documentation for esp8266. esp32 didn't have good docs then. so here it is 
https://docs.espressif.com/projects/arduino-esp32/en/latest/api/wifi.html#begin

Playit3110, 2023-01-14:
I see, so i can just use the standart begin method to connect with the bssid of an AP. Thank you for your help

JAndrassy, 2023-01-14:
[image]