JAndrassy/WiFiEspAT #104: Arduino Nano, ESP-1, and WiFiSSLClient

dickhg, 2023-05-15:
About a year and a half ago I developed a Nano/ESP-01 project in which visited 3 websites, collected data, and displayed on UTC time, weather, and some solar data on a 4 LCD line display.  It worked great and was very useful.  A few weeks ago, I quit receiving the solar data and I figured out that the web source I was using had changed its protocols and was now requiring a SSL connection.  I tried moving from the WiFiClient to the WiFiSSLClient and, of course, realized that I needed new 8266 firmware.  I beat my head against the wall on this problem until I found ESP-1MB-tx1rx3-AT_V2.2 flash code a few days ago.  It flashes with this message:

AT version:2.2.0.0(s-b097cdf - ESP8266 - Jun 17 2021 12:58:29)
SDK version:v3.4-22-g967752e2
compile time(6800286):Jul 11 2021 11:09:32
Bin version:2.2.0(ESP8266_1MB)

I then tried this newly flashed ESP-01 in my project and it wouldn't connect to my WiFi router.  Same result with the WiFiSSLClient example.  

The code gets past communicating with WiFi module (it knows the ESP-01 is there), but never passes the WL_CONECTED test.  I put in a print command and the first status code received back is 4 and then after every pass after that is 2.

I'm about ready to start this project over with an ESP32 that doesn't require the firmware to be shoed-horned in, but before I do that, I wondered if there is something I'm missing.
Thanks.

JAndrassy, 2023-05-16:
I recommend you to read the README file

dickhg, 2023-05-17:
I've tried Jiri Bilek's firmware, but can't get it compiled either through the Arduino IDE or Platform.io.    I think that I have to accept the fact that I'm probably too far over my head on this and can't figure it out without asking for an unreasonable amount of help.  So, it's time to drop back and try something else.  Thanks for your help.

JAndrassy, 2023-05-18:
> I've tried Jiri Bilek's firmware, but can't get it compiled either through the Arduino IDE
use esp8266 arduino version 2.7.4

dickhg, 2023-05-24:
Thanks for help, but no luck here.  I'm moving on to a part with more memory...