JAndrassy/WiFiEspAT #108: WiFiWebServer.h: No such file or directory

hilmiyafia, 2023-06-10:
I tried running the HelloServer example on my Arduino Mega, but this error came up at compilation:

WiFiWebServer.h: No such file or directory

JAndrassy, 2023-06-10:
sorry, then I thought the WiFiWebServer library would become well known. 
[image]

hilmiyafia, 2023-06-10:
@JAndrassy Thank you for the reply, but now I have this error:

[image]

JAndrassy, 2023-06-11:
it was working on Mega then. now it looks like there is some incompatibility in the WiFiWebServer library. try older versions of the  WiFiWebServer library

hilmiyafia, 2023-06-12:
@JAndrassy Ok thank you, the latest version that works is 1.4.2 😊

closed 2023-06-12 by hilmiyafia