JAndrassy/WiFiEspAT #109: Arduino Uno ATmega328P + ESP8266 + AT 1.7.5 + WiFiEspAT + OTA

brunoaduarte, 2023-07-11:
Hi,

Is it possible to implement OTA updates to the ATmega328P (Arduino Uno) sketch on this board using this library?

Are there any examples on how to do that?

Seems the ArduinoOTA library is not compatible with the ATmega328P.

I've already installed the AT 1.7.5 firmware on the ESP8266 following this tutorial
https://www.sigmdel.ca/michel/ha/esp8266/ESP01_AT_Firmware_en.html

And also connected the software serial pins 6,7 to the corresponding ESP TXD,RXD pins.

Tested the communication and it's ok.

[image]
[image]

JAndrassy, 2023-07-11:
> Is it possible to implement OTA updates to the ATmega328P (Arduino Uno) sketch on this board using this library?

not. there is not enough flash memory for self-update of Uno

more here https://github.com/JAndrassy/ArduinoOTA#atmega-support

brunoaduarte, 2023-07-11:
> > Is it possible to implement OTA updates to the ATmega328P (Arduino Uno) sketch on this board using this library?
> 
> not. there is not enough flash memory for self-update of Uno
> 
> more here https://github.com/JAndrassy/ArduinoOTA#atmega-support

Are there any other library you recommend that will make it possible?
As by your comment here
https://forum.arduino.cc/t/robotdyn-clone-uno-wifi-esp8266-board-firmware-and-ota/688616/8
"3.) there are esp8266 firmware which can OTA update ATmega328p"

JAndrassy, 2023-07-11:
I repeat
more here https://github.com/JAndrassy/ArduinoOTA#atmega-support

closed 2023-10-19 by JAndrassy