JAndrassy/WiFiEspAT #110: BlynkSimpleWifi.h does not exist?

Thereal14thDoctor, 2023-07-13:
I couldn't find another way to ask you this, but in your blynk wifi examples you have included a library called BlynkSimpleWifi.h.  This library does not exist in the library manager.  Is there a replacement for this library?

Thereal14thDoctor, 2023-07-13:
Ok, this is what I get from the serial monitor.  does this mean the firmware was not flashed correctly?

[image]

Thereal14thDoctor, 2023-07-13:
Sorry, forgot to change the dips back.  now I get alot of info:
12:57:50.591 -> esp INFO: soft reset
12:57:50.591 -> esp> AT+RST ...sent
12:57:50.591 -> esp> AT+RST ...ignored
12:57:50.591 -> esp> OK ...ignored
12:57:50.685 -> esp>  ets Jan  8 2013,rst cause:2, boot mode:(3,6) ...ignored
12:57:50.685 -> esp> load 0x40100000, len 2592, room 16  ...ignored
12:57:50.731 -> esp> tail 0 ...ignored
12:57:50.731 -> esp> chksum 0xf3 ...ignored
12:57:50.731 -> esp> load 0x3ffe8000, len 764, room 8  ...ignored
12:57:50.731 -> esp> tail 4 ...ignored
12:57:50.731 -> esp> chksum 0x92 ...ignored
12:57:50.731 -> esp> load 0x3ffe82fc, len 676, room 4  ...ignored
12:57:50.731 -> esp> tail 0 ...ignored
12:57:50.731 -> esp> chksum 0x22 ...ignored
12:57:50.731 -> esp> csum 0x22 ...ignored
12:57:50.731 -> esp> 2nd boot version : 1.7(5d6f877) ...ignored
12:57:50.731 -> esp> SPI 40MHz ...ignored
12:57:50.731 -> esp> SPI Moash Size & 4KB+1024KB) ...ignored
12:57:50.772 -> esp> jump to run user1 @ 10correct flash map ...ignored
12:57:50.772 -> esp> sn|s⸮rnln⸮⸮l⸮b|⸮n⸮|bc⸮⸮o⸮⸮⸮⸮l⸮r⸮⸮⸮nb⸮ob⸮n⸮⸮~⸮ol⸮ ...ignored
12:57:50.888 -> esp> |⸮⸮n|rl⸮s⸮⸮⸮ob⸮n⸮⸮~⸮o⸮⸮⸮o⸮⸮c⸮⸮o|⸮⸮⸮⸮b⸮ol⸮bl`r⸮olph⸮⸮ ...ignored
12:57:50.888 -> esp> ⸮⸮o⸮r⸮⸮o|⸮l⸮⸮lb⸮⸮|s⸮l⸮n⸮⸮n⸮ ...ignored
12:57:50.888 -> esp> ⸮l⸮l⸮l`⸮⸮r⸮l⸮⸮ ...ignored
12:57:50.888 -> esp> on⸮⸮lp⸮l⸮⸮ll⸮l⸮⸮l⸮l⸮n⸮ ...ignored
12:57:50.888 -> esp> bslrl ...ignored
12:57:50.888 -> esp> ready ...matched
12:57:50.888 -> esp> ATE0 ...sent
12:57:50.888 -> esp> ATE0 ...ignored
12:57:50.888 -> esp> OK ...matched
12:57:50.888 -> esp> AT+CIPMUX=1 ...sent
12:57:50.888 -> esp> OK ...matched
12:57:50.888 -> esp> AT+CIPRECVMODE=1 ...sent
12:57:50.888 -> esp> OK ...matched
12:57:50.888 -> esp> AT+CWMODE? ...sent
12:57:50.888 -> esp> +CWMODE:1 ...matched
12:57:50.888 -> esp> OK ...matched
12:57:50.888 -> esp INFO: wifi status
12:57:50.888 -> esp> AT+CIPSTATUS ...sent
12:57:50.888 -> esp> STATUS:5 ...matched
12:57:50.888 -> esp> OK ...matched
12:57:50.888 -> esp INFO: quit AP  current
12:57:50.888 -> esp> AT+CIPDNS_CUR=0 ...sent
12:57:50.888 -> esp> OK ...matched
12:57:50.888 -> esp> AT+CWDHCP_CUR=1,1 ...sent
12:57:50.888 -> esp> OK ...matched
12:57:50.888 -> esp> AT+CWQAP ...sent
12:57:50.888 -> esp> OK ...matched
12:57:50.888 -> esp INFO: end SoftAP persistent
12:57:50.888 -> 
12:57:50.888 -> Attempting to connect to SSID: SpectrumSetup-67
12:57:50.912 -> 
12:57:50.912 -> Pass
12:57:50.912 -> energymonth147
12:57:50.912 -> esp INFO: join AP SpectrumSetup-67 persistent
12:57:50.912 -> esp> AT+CWJAP="SpectrumSetup-67","energymonth147" ...sent
12:57:51.928 -> esp> ?
12:57:51.928 -> esp> busy p... ...ignored
12:57:52.911 -> esp> ?
12:57:52.911 -> esp> busy p... ...ignored
12:57:53.802 -> esp> WIFI CONNECTED ...ignored
12:57:54.780 -> esp> ?
12:57:54.780 -> esp> busy p... ...ignored
12:57:55.784 -> esp> ?
12:57:55.784 -> esp> busy p... ...ignored
12:57:56.803 -> esp> ?
12:57:56.803 -> esp> busy p... ...ignored
12:57:57.796 -> esp> ?
12:57:57.796 -> esp> busy p... ...ignored
12:57:58.807 -> esp> ?
12:57:58.807 -> esp> busy p... ...ignored
12:57:59.827 -> esp> ?
12:57:59.827 -> esp> busy p... ...ignored
12:58:00.830 -> esp> ?
12:58:00.830 -> esp> busy p... ...ignored
12:58:01.818 -> esp> ?
12:58:01.818 -> esp> busy p... ...ignored
12:58:02.810 -> esp> ?
12:58:02.810 -> esp> busy p... ...ignored
12:58:03.788 -> esp> ?
12:58:03.788 -> esp> busy p... ...ignored
12:58:04.806 -> esp> ?
12:58:04.806 -> esp> busy p... ...ignored
12:58:04.901 -> esp> +CWJAP:1 ...ignored
12:58:04.901 -> esp> FAIL ...error
12:58:04.901 -> esp ERROR: expected OK got FAIL
12:58:04.947 -> esp INFO: quit AP  persistent
12:58:04.947 -> esp> AT+CWAUTOCONN=0 ...sent
12:58:04.947 -> esp> OK ...matched
12:58:04.947 -> esp> AT+CIPDNS_DEF=0 ...sent
12:58:04.947 -> esp> WIFI DISCONNECT ...ignored
12:58:04.947 -> esp> OK ...matched
12:58:04.947 -> esp> AT+CWDHCP=1,1 ...sent
12:58:04.947 -> esp> OK ...matched
12:58:04.947 -> esp> AT+CWQAP ...sent
12:58:04.947 -> esp> OK ...matched
12:58:04.947 -> 
12:58:04.947 -> Connection to WiFi network failed.

Thereal14thDoctor, 2023-07-13:
I am so sorry to be so much trouble, but I am learning quite a bit.  I wish there were more amazing people like you.

Thereal14thDoctor, 2023-07-13:
Sorry, forgot to change the dips back. now I get alot of info:
12:57:50.591 -> esp INFO: soft reset
12:57:50.591 -> esp> AT+RST ...sent
12:57:50.591 -> esp> AT+RST ...ignored
12:57:50.591 -> esp> OK ...ignored
12:57:50.685 -> esp>  ets Jan  8 2013,rst cause:2, boot mode:(3,6) ...ignored
12:57:50.685 -> esp> load 0x40100000, len 2592, room 16  ...ignored
12:57:50.731 -> esp> tail 0 ...ignored
12:57:50.731 -> esp> chksum 0xf3 ...ignored
12:57:50.731 -> esp> load 0x3ffe8000, len 764, room 8  ...ignored
12:57:50.731 -> esp> tail 4 ...ignored
12:57:50.731 -> esp> chksum 0x92 ...ignored
12:57:50.731 -> esp> load 0x3ffe82fc, len 676, room 4  ...ignored
12:57:50.731 -> esp> tail 0 ...ignored
12:57:50.731 -> esp> chksum 0x22 ...ignored
12:57:50.731 -> esp> csum 0x22 ...ignored
12:57:50.731 -> esp> 2nd boot version : 1.7(5d6f877) ...ignored
12:57:50.731 -> esp> SPI 40MHz ...ignored
12:57:50.731 -> esp> SPI Moash Size & 4KB+1024KB) ...ignored
12:57:50.772 -> esp> jump to run user1 @ 10correct flash map ...ignored
12:57:50.772 -> esp> sn|s⸮rnln⸮⸮l⸮b|⸮n⸮|bc⸮⸮o⸮⸮⸮⸮l⸮r⸮⸮⸮nb⸮ob⸮n⸮⸮~⸮ol⸮ ...ignored
12:57:50.888 -> esp> |⸮⸮n|rl⸮s⸮⸮⸮ob⸮n⸮⸮~⸮o⸮⸮⸮o⸮⸮c⸮⸮o|⸮⸮⸮⸮b⸮ol⸮bl`r⸮olph⸮⸮ ...ignored
12:57:50.888 -> esp> ⸮⸮o⸮r⸮⸮o|⸮l⸮⸮lb⸮⸮|s⸮l⸮n⸮⸮n⸮ ...ignored
12:57:50.888 -> esp> ⸮l⸮l⸮l`⸮⸮r⸮l⸮⸮ ...ignored
12:57:50.888 -> esp> on⸮⸮lp⸮l⸮⸮ll⸮l⸮⸮l⸮l⸮n⸮ ...ignored
12:57:50.888 -> esp> bslrl ...ignored
12:57:50.888 -> esp> ready ...matched
12:57:50.888 -> esp> ATE0 ...sent
12:57:50.888 -> esp> ATE0 ...ignored
12:57:50.888 -> esp> OK ...matched
12:57:50.888 -> esp> AT+CIPMUX=1 ...sent
12:57:50.888 -> esp> OK ...matched
12:57:50.888 -> esp> AT+CIPRECVMODE=1 ...sent
12:57:50.888 -> esp> OK ...matched
12:57:50.888 -> esp> AT+CWMODE? ...sent
12:57:50.888 -> esp> +CWMODE:1 ...matched
12:57:50.888 -> esp> OK ...matched
12:57:50.888 -> esp INFO: wifi status
12:57:50.888 -> esp> AT+CIPSTATUS ...sent
12:57:50.888 -> esp> STATUS:5 ...matched
12:57:50.888 -> esp> OK ...matched
12:57:50.888 -> esp INFO: quit AP  current
12:57:50.888 -> esp> AT+CIPDNS_CUR=0 ...sent
12:57:50.888 -> esp> OK ...matched
12:57:50.888 -> esp> AT+CWDHCP_CUR=1,1 ...sent
12:57:50.888 -> esp> OK ...matched
12:57:50.888 -> esp> AT+CWQAP ...sent
12:57:50.888 -> esp> OK ...matched
12:57:50.888 -> esp INFO: end SoftAP persistent
12:57:50.888 -> 
12:57:50.888 -> Attempting to connect to SSID: SpectrumSetup-67
12:57:50.912 -> 
12:57:50.912 -> Pass
12:57:50.912 -> xxxxxxxx (password)
12:57:50.912 -> esp INFO: join AP SpectrumSetup-67 persistent
12:57:50.912 -> esp> AT+CWJAP="SpectrumSetup-67","xxxxxxx" ...sent
12:57:51.928 -> esp> ?
12:57:51.928 -> esp> busy p... ...ignored
12:57:52.911 -> esp> ?
12:57:52.911 -> esp> busy p... ...ignored
12:57:53.802 -> esp> WIFI CONNECTED ...ignored
12:57:54.780 -> esp> ?
12:57:54.780 -> esp> busy p... ...ignored
12:57:55.784 -> esp> ?
12:57:55.784 -> esp> busy p... ...ignored
12:57:56.803 -> esp> ?
12:57:56.803 -> esp> busy p... ...ignored
12:57:57.796 -> esp> ?
12:57:57.796 -> esp> busy p... ...ignored
12:57:58.807 -> esp> ?
12:57:58.807 -> esp> busy p... ...ignored
12:57:59.827 -> esp> ?
12:57:59.827 -> esp> busy p... ...ignored
12:58:00.830 -> esp> ?
12:58:00.830 -> esp> busy p... ...ignored
12:58:01.818 -> esp> ?
12:58:01.818 -> esp> busy p... ...ignored
12:58:02.810 -> esp> ?
12:58:02.810 -> esp> busy p... ...ignored
12:58:03.788 -> esp> ?
12:58:03.788 -> esp> busy p... ...ignored
12:58:04.806 -> esp> ?
12:58:04.806 -> esp> busy p... ...ignored
12:58:04.901 -> esp> +CWJAP:1 ...ignored
12:58:04.901 -> esp> FAIL ...error
12:58:04.901 -> esp ERROR: expected OK got FAIL
12:58:04.947 -> esp INFO: quit AP  persistent
12:58:04.947 -> esp> AT+CWAUTOCONN=0 ...sent
12:58:04.947 -> esp> OK ...matched
12:58:04.947 -> esp> AT+CIPDNS_DEF=0 ...sent
12:58:04.947 -> esp> WIFI DISCONNECT ...ignored
12:58:04.947 -> esp> OK ...matched
12:58:04.947 -> esp> AT+CWDHCP=1,1 ...sent
12:58:04.947 -> esp> OK ...matched
12:58:04.947 -> esp> AT+CWQAP ...sent
12:58:04.947 -> esp> OK ...matched
12:58:04.947 -> 
12:58:04.947 -> Connection to WiFi network failed.

JAndrassy, 2023-07-14:
the esp8266 connects to WiFi, but doesn't receive an IP address over DHCP.
the return code is 1 (+CWJAP:1), which is timeout.
is the signal level good? you can run the ScanNetworksAdvanced sketch to list APs and RSSI

Thereal14thDoctor, 2023-07-14:
ScanNetworksAdvanced returns:
09:45:06.181 -> Scanning available networks...
09:45:06.181 -> ** Scan Networks **
09:45:06.181 -> esp INFO: list AP
09:45:06.181 -> esp> AT+CWLAPOPT=1,31 ...sent
09:45:06.181 -> esp> OK ...matched
09:45:06.181 -> esp> AT+CWLAP ...sent
09:45:07.163 -> esp> ?
09:45:07.163 -> esp> busy p... ...ignored
09:45:08.198 -> esp> ?
09:45:08.198 -> esp> busy p... ...ignored
09:45:08.371 -> esp> +CWLAP:(3,"SpectrumSetup-67",-28,"b0:fc:88:67:55:09",11) ...matched
09:45:08.371 -> esp> +CWLAP:(4,"Alula0390D4",-70,"54:21:60:03:90:d5",1) ...matched
09:45:08.371 -> esp> +CWLAP:(3,"july41959",-74,"28:80:88:f8:ce:31",1) ...matched
09:45:08.371 -> esp> +CWLAP:(3,"Paul Lainey",-87,"14:7d:05:ea:2d:2f",1) ...matched
09:45:08.371 -> esp> +CWLAP:(3,"ATT7ap3Qge",-89,"dc:7f:a4:d3:a1:29",4) ...matched
09:45:08.419 -> esp> +CWLAP:(3,"july41959",-89,"40:2b:50:59:69:20",5) ...matched
09:45:08.419 -> esp> +CWLAP:(3,"AT,-91,"c0:89:ab:04:0c:d0",1) ...ignored
09:45:08.419 -> esp> +CWLAP:(0,"SmartLife-4:32:d0:e3:29",1) ...ignored
09:45:08.419 -> esp> +CWLAP:(3,"SpectrumSetu"f0:81:75:d5:b4:46",1) ...ignored
09:45:08.419 -> esp> OK ...matched
09:45:08.419 -> number of available networks: 6
09:45:08.419 -> 1) Signal: -28 dBm	Channel: 11		BSSID: B0:FC:88:67:55:09
09:45:08.419 -> 	Encryption: WPA2		SSID: SpectrumSetup-67
09:45:08.419 -> 2) Signal: -70 dBm	Channel: 1		BSSID: 54:21:60:03:90:D5
09:45:08.419 -> 	Encryption: WPA2		SSID: Alula0390D4
09:45:08.419 -> 3) Signal: -74 dBm	Channel: 1		BSSID: 28:80:88:F8:CE:31
09:45:08.465 -> 	Encryption: WPA2		SSID: july41959
09:45:08.465 -> 4) Signal: -87 dBm	Channel: 1		BSSID: 14:7D:05:EA:2D:2F
09:45:08.465 -> 	Encryption: WPA2		SSID: Paul Lainey
09:45:08.465 -> 5) Signal: -89 dBm	Channel: 4		BSSID: DC:7F:A4:D3:A1:29
09:45:08.465 -> 	Encryption: WPA2		SSID: ATT7ap3Qge
09:45:08.465 -> 6) Signal: -89 dBm	Channel: 5		BSSID: 40:2B:50:59:69:20
09:45:08.465 -> 	Encryption: WPA2		SSID: july41959
I don't know what to make of the numbers.  "Signal: -28"
is this a very low signal?  I am literally 6 feet from the router.

Thereal14thDoctor, 2023-07-14:
Well, it is finally working.  I had to manually assign the IP address in the setuppersistentwificonnection sketch.  ran it again and it connected.  reloaded the Blynk sketch and finally connected.  Thank you so much for all of your help.  You really are amazing and I count myself lucky to have found you here.  I wish you a wonderful weekend, try to stay cool.  104 F here today, and tomorrow, and the next day... Please know that you are appreciated so much!

Thereal14thDoctor, 2023-07-20:
I really thought this was working but apparently I am making a basic mistake somewhere but I cannot find it.  The board(mega2560) does connect according to the serial monitor.  The mobile Blynk app connects and I can tell that it is communicating as the Serial Monitor shows:
12:39:17.593 -> esp> +IPD,4,11 ...processed
12:39:17.593 -> esp INFO: get data on link 4
12:39:17.593 -> esp> AT+CIPRECVDATA=4,64 ...sent
12:39:17.593 -> esp> +CIPRECVDATA,11 ...matched
12:39:17.639 -> esp> OK ...matched
12:39:17.639 -> esp INFO: 	got 11 bytes on link 4
12:39:17.639 -> esp INFO: sync
12:39:17.639 -> esp> AT+CIPSTATUS ...sent
12:39:17.639 -> esp> STATUS:3 ...matched
12:39:17.685 -> esp> +CIPSTATUS:4,"TCP","64.225.16.22",80,3912,0 ...matched
12:39:17.685 -> esp> OK ...end of list
when the button on the Blynk App is pressed, but I cannot get the board to turn the led on.  Led is on pin 8. 
This is the only thing connected to the arduino.
here is my code:
/*************************************************************
 

  Blynk library is licensed under MIT license
  This example code is in public domain.

 modified in Jul 2019 for WiFiEspAT library
 by Juraj Andrassy https://github.com/jandrassy

 *************************************************************/

/* Comment this out to disable prints and save space */

#define BLYNK_TEMPLATE_ID           "TMPL2-wecEaRx"
#define BLYNK_TEMPLATE_NAME         "Quickstart Template"
#define BLYNK_AUTH_TOKEN            "2OjXNvcKckOZ1fc7mDYWS3UfA7pH9x_S"
#define BLYNK_PRINT Serial

#include <WiFiEspAT.h>
#include <BlynkSimpleWifi.h>

// Your WiFi credentials.
// Set password to "" for open networks.
char ssid[] = "xxxxxxx";
char pass[] = "xxxxx";

// Emulate Serial1 on pins 6/7 if not present
#if defined(ARDUINO_ARCH_AVR) && !defined(HAVE_HWSERIAL1)
#include <SoftwareSerial.h>
SoftwareSerial Serial1(6, 7); // RX, TX
#define AT_BAUD_RATE 9600
#else
#define AT_BAUD_RATE 115200
#endif

// You should get Auth Token in the Blynk App.
// Go to the Project Settings (nut icon).
char auth[] = "2OjXNvcKckOZ1fc7mDYWS3UfA7pH9x_S";

BLYNK_WRITE(V0)
{
  int pinValue = param.asInt();
  digitalWrite(8, pinValue);
  
}

void setup() 
{
  pinMode(8, OUTPUT);
  
  
  
  Serial.begin(115200);
  while (!Serial);

  Serial3.begin(AT_BAUD_RATE);
  WiFi.init(Serial3);

  if (WiFi.status() == WL_NO_MODULE) {
    Serial.println("Communication with WiFi module failed!");
    // don't continue
    while (true);
  }

  Blynk.begin(BLYNK_AUTH_TOKEN, ssid, pass);

  // waiting for connection to Wifi network set with the SetupWiFiConnection sketch
  Serial.println("Waiting for connection to WiFi");
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.print('.');
  }
  Serial.println("You're connected to the network");

  Blynk.config(auth, BLYNK_DEFAULT_DOMAIN, BLYNK_DEFAULT_PORT);
  while(Blynk.connect() != true) {}
}

void loop()
{
  Blynk.run();
}
I am very grateful for all the assistance.

Thereal14thDoctor, 2023-07-20:
Very Sorry, and equally embarrassed.  There is nothing wrong with the code, it all works.  As long as the led is connected to digital pin 8 instead of analog 8.  Please laugh at me as I laugh at myself for making such a basic beginner mistake.  I promise I will diagnose my project issues more thoroughly in the future before asking for assistance.  Thank you again for all of your patience and assistance.  Be Well.

closed 2023-07-20 by Thereal14thDoctor