JAndrassy/WiFiEspAT #111: WifiClient has no setConnectionTimeout

lomithrani, 2023-09-03:
When trying to connect to a server that doesn't exist or isn't responding connect hangs.
It would be nice to have an option to setConnectionTimeout.

JAndrassy, 2023-09-03:
there is no AT command or parameter to set a connect timeout and the library must always wait until the AT firmware finishes the command with OK or ERROR.

closed 2023-09-03 by JAndrassy

lomithrani, 2023-09-03:
@JAndrassy thanks for the answer ? Could you elaborate why this should be a requirement ? For instance I'm using WifiClient with HttpClient arduino and every 10 seconds I post some data. If for some reason the server is down I can't have my connect function holding my program for 30 (?) seconds

JAndrassy, 2023-09-04:
because until the AT firmware didn't finish the execution of an AT command, it will not accept any other command. there is no way to abort execution of the current command.  
it would require additional state and logic to recover from unknown state of then communication with the AT firmware

JAndrassy, 2023-09-05:
you could use WiFi.ping to check if the IP is up

lomithrani, 2023-09-30:
@JAndrassy Doesnt't the Maintain() in Wifi.ping  which calls `EspAtDrvClass::readRX` result in the exact same issue ? 
I've added a simple :
```cpp
    currentMillis = millis();

    if (timeoutOn && (currentMillis - startMillis >= 1000)) // TODO REMOVE THIS ATROCITY
    {
      return false;
    }
```
In the while loop of `EspAtDrvClass::readRX` for now. If I find time to do it properly with recovery I'll hit you up with a PR !

JAndrassy, 2023-09-30:
I would expect ping to return faster than connect if the server is not reachable.

readRX is where it waits. there is a timeout, but if that timeout happens, next library function calls may fail too.