JAndrassy/WiFiEspAT #113: Communication with WiFi module failed!

JEAPI-DEV, 2023-11-23:
Hey

I'm using this module: https://www.seeedstudio.com/Grove-UART-WiFi-V2-ESP8285.html. I know the chip (ESP8285) isn't officially supported by this library. Any tips on making it work with this module?

When printing out the WiFi.status, I get the value 255. Maybe that can tell you something, maybe not.

Thanks!

JEAPI-DEV, 2023-11-24:
Hi!

I checked with enabled debug, and here's the output:

```
esp INFO: soft reset
esp> AT+RST ...sent
esp INFO: soft reset
esp> AT+RST ...sent
esp> ?
esp> ?
esp> ?
esp ERROR: AT firmware not responding
esp> ATE0 ...sent
esp> ?
esp> ?
esp> ?
esp ERROR: AT firmware not responding
```

Thanks for your response btw.

JAndrassy, 2023-11-24:
https://github.com/JAndrassy/WiFiEspAT#getting-started

JEAPI-DEV, 2023-11-24:
Would it be possible to update the firmware to 1.7?

JEAPI-DEV, 2023-11-24:
And sorry seems like you documented everything perfectly and I'm just asking questions you already answered.

closed 2024-08-03 by JAndrassy