JAndrassy/WiFiEspAT #114: Not connecting to a mobile hotspot

itsosmx, 2024-01-14:
Hello, I'm opennig a mobile hotspot to connect the esp8266 to it, but its not working i dont know why? the problem also its print that iam connecting when iam not 
```
15:46:13.464 -> [Arduino] Attempting to connect to WPA SSID:  IOTM 
15:46:19.305 -> [Arduino] SSID:  IOTM 
15:46:19.305 -> [Arduino] IP Address: 192.168.1.111
```
the connection to the home network working well and everything working prefect the problem with the hotspot 
[image]
[image]
ذذذ
```cpp
bool InitESP() {
  WiFi.init(Serial1);  // initialize ESP module
  WiFi.config(ip);
  // check for the presence of the shield
  if (WiFi.status() == WL_NO_MODULE) {
    Print("Communication with WiFi module failed!");
    return false;
  } else {
    // attempt to connect to WiFi network
    while (status != WL_CONNECTED) {
      Print("Attempting to connect to WPA SSID: ", ssid);
      // Connect to WPA/WPA2 network
      status = WiFi.begin(ssid, pass);
      delay(1000);
    }
    Print("SSID: ", WiFi.SSID());
    Serial.print("[Arduino] IP Address: ");
    Serial.println(WiFi.localIP());
    return true;
  }
}
```

JAndrassy, 2024-01-14:
ESP chips don't support 5 GHz WiFi

itsosmx, 2024-01-14:
@JAndrassy I changed the band to 2.4GHz and also not working.

itsosmx, 2024-01-14:
more details 
```
18:35:52.142 -> esp> AT+CIPSTATUS ...sent
18:35:52.142 -> esp> STATUS:5 ...matched
18:35:52.177 -> esp> OK ...matched
18:35:52.177 -> [Arduino] Attempting to connect to WPA SSID:  IOTM 
18:35:52.177 -> esp INFO: join AP IOTM current
18:35:52.177 -> esp> AT+CWJAP_CUR="IOTM","60606060" ...sent
18:35:53.166 -> esp> ?
18:35:53.213 -> esp> busy p... ...ignored
18:35:54.198 -> esp> ?
18:35:54.234 -> esp> busy p... ...ignored
18:35:55.103 -> esp> WIFI CONNECTED ...ignored
18:35:55.149 -> esp> WIFI GOT IP ...ignored
18:35:55.228 -> esp> OK ...matched
18:35:56.235 -> esp INFO: AP query
18:35:56.235 -> esp> AT+CWJAP? ...sent
18:35:56.269 -> esp> +CWJAP:"IOTM","72:1f:3c:54:17:af",1,-37,0 ...matched
18:35:56.269 -> esp> OK ...matched
18:35:56.269 -> [Arduino] SSID:  IOTM 
18:35:56.269 -> [Arduino] IP Address: esp INFO: STA IP query
18:35:56.269 -> esp> AT+CIPSTA? ...sent
18:35:56.340 -> esp> +CIPSTA:ip:"192.168.1.111" ...matched
18:35:56.374 -> esp> +CIPSTA:gateway:"192.168.1.1" ...matched
18:35:56.407 -> esp> +CIPSTA:netmask:"255.255.255.0" ...matched
18:35:56.407 -> esp> OK ...matched
18:35:56.407 -> 192.168.1.111
18:35:56.407 -> esp INFO: begin server at port 80
18:35:56.408 -> esp> AT+CIPSERVERMAXCONN=1 ...sent
18:35:56.442 -> esp> OK ...matched
18:35:56.442 -> esp> AT+CIPSERVER=1,80 ...sent
18:35:56.484 -> esp> OK ...matched
18:35:56.484 -> esp> AT+CIPSTO=60 ...sent
18:35:56.484 -> esp> OK ...matched
18:35:56.484 -> [Arduino] Done Initialize 
```

JAndrassy, 2024-01-14:
a server? I think a hotspot only allows to connect to Internet over the 'data' connection. It will not allow the connected devices to communicate to each other and the phone itself can't access the devices too. It is not a router.

itsosmx, 2024-01-14:
oh no do you know any workaround or something to make it work with hotspot , i need to make it cuz it will work in a place where there is no wifi

JAndrassy, 2024-01-15:
to access the Arduino from phone you can start local Access Point on the Arduino

itsosmx, 2024-01-15:
Its working now, after removeing the config ip
```cpp
WiFi.config(ip)
```

closed 2024-01-15 by itsosmx