JAndrassy/WiFiEspAT #115: Not connecting to Wifi with AT version:2.2.2.0

alexlopez2612, 2024-03-10:
Hello

I'm Alexis, nice to meet you. 
First, thanks for this library, it's very powerfull and simply to understand.
I'm new on this and I have a Arduino Mega with ESP8266 onboard. 
At first, I update the firmware to:

AT version:1.7.5.0(Oct  9 2021 09:26:04)
SDK version:3.0.5(b29dcd3)
compile time:Oct 15 2021 18:05:38
Bin version(Wroom 02):1.7.5

And with your examples, I could run a web service and the WebClient.ino, working ok
But, when I tried to connect to https web service, to get weather info, with the WiFiSSLClient.ino, I get no response from the web service.
Reading over the internet, I saw that, I think you, this will not work with At versions lower than 2.1.
So, after a lot of searching and time, I could update the 8266 to:

AT version:2.2.2.0-dev(952f658 - ESP8266 - Nov 23 2022 06:37:28)
SDK version:v3.4-73-g4fd37f89
compile time(8261016):Feb 26 2024 13:11:08
Bin version:2.2.1(ESP8266_1MB)

But, with the same sketch, I cant connect to wifi.

I don't know if Im doing something wrong, but is the same program that works diferent. But, I thought to tell you because maybe it´s a bug that you don't know yet.

This is the code:
`#include <WiFiEspAT.h>
//Variables WIFI
#define NAME_OF_SSID "LinsiNet"
#define PASSWORD_OF_SSID "CasaAlexito"

WiFiSSLClient client;

void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial3.begin(115200);
  WiFi.init(Serial3);

  if (WiFi.status() == WL_NO_MODULE) {
    Serial.println();
    Serial.println("Communication with WiFi module failed!");
    // don't continue
    while (true);
  }
 WiFi.begin(NAME_OF_SSID, PASSWORD_OF_SSID);
  // waiting for connection to Wifi network set with the SetupWiFiConnection sketch
  Serial.println("Waiting for connection to WiFi");
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.print('.');
  }
  Serial.println();
  Serial.println("Connected to WiFi network.");

const char* server = "api.github.com1";
//const char* server = "el-tiempo.net";

  Serial.println("Starting connection to server...");
  if (client.connect(server, 443)) { // port 443 is the default https port
    Serial.println("connected to server");

    //client.println("GET /api/json/v2/provincias/28/municipios/28903 HTTP/1.1");
    client.println("GET /repos/jandrassy/WiFiEspAT/commits/master/status HTTP/1.1");
    client.print("Host: ");
    client.println(server);
    client.println("User-Agent: Arduino");
    client.println("Connection: close");
    client.println();
    client.flush();
  }
}`

It's my first time on github, so I'm really sorry if I'm doing this wrong.

I hope you can tell me something.

Best Regards and thanks in advance

JAndrassy, 2024-03-10:
I recommend you to use the 'fake' AT firmware from Jiri Bilek

alexlopez2612, 2024-03-11:
> I recommend you to use the 'fake' AT firmware from Jiri Bilek

Hello, thanks for your reply,

I tried that, and It works the same that the version 1.7.5. Connects ok to wifi, but I get no data from the web service

This is the version of the Jiri's firmware:
AT version:1.7.0.0 (partial)
SDK version:2.2.2-dev(38a443e)
Compile time:Mar 11 2024 15:41:04
Version ESP_ATMod:0.4.0beta
Arduino core version:3.1.2

am I doing anything wrong???

Thanks!!!!

JAndrassy, 2024-03-11:
is the 1 "api.github.com1" in your test?

alexlopez2612, 2024-03-11:
Oh my god....

That's there because I tried to corrupt the URL to see if I get anything... I fix that and works ok. Thank you very much

To summary, for your information, the ssl client works fine with Jiri Bilek's AT firmware.
One thing, works with the one ESP_ATMod-Core_3, because ESP_ATMod is not compilling. He has an issue of that.

Thank you very much for yor reply and for the library, it's really easy to understand for people with not that much experience.

Best regards

closed 2024-03-11 by JAndrassy